pimoroni/pimoroni-pico #470: Badger 2040 firmware without the python "badger os" examples

TheLogster, 2022-07-26:
It would be great to have the option to download a build of the firmware for the badger 2040 device, that doesn't have the baked in python examples.

alphanumeric007, 2022-07-26:
You could try the basic Pimoroni Pico uf2, it has all the libraries in it for the Pimoroni products.

Or just delete the examples. In Thonny go to "View" and select "Files". You then get a list of what is stored on the device.

TheLogster, 2022-07-27:
>You could try the basic Pimoroni Pico uf2, it has all the libraries in it for the Pimoroni products.

The pimoroni-pico-v1.19.5-micropython.uf2 doesn't contain the badger2040 module

>Or just delete the examples.

The examples are "baked" into the firmware via micropython-builtins.cmake in the micropython/examples/badger2040 folder.

alphanumeric007, 2022-07-27:
I could have sworn I had replied to your last post? Must have hit the wrong button?

I meant delete them from the Badger, after you flash it. Sorry for the confusion, could have worded it better.

It's what I did with my Tufty. Then I made my own main.py to run what I wanted on boot up. I have also just left the original main.py and aded my py files to my Tufty. The stock main py finds them and adds them to its list. I can then run the one I want from the list. 
My Tufty appears to be a lot simpler in operation than your Badger though. I could get it up and running with just the stock Pico uf2 file. 

You may be getting your wish though, have a look see here, scroll down the list.
https://github.com/pimoroni/pimoroni-pico/actions

Gadgetoid, 2022-07-27:
Hahaha good spot :laughing:

Gadgetoid, 2022-07-27:
You can grab a "without BadgerOS" firmware here - https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.6

closed 2022-07-27 by Gadgetoid