pimoroni/pimoroni-pico #472: v1.19.5 Pico W uf2 firmware is missing urequests

TheLogster, 2022-07-27:
The following code works on a stock Rasberry pico w firmware, but on v.1.19.5 pimoroni pico w firmare (pimoroni-picow-v1.19.5-micropython.uf2) an import error occurs:

`import network
import secrets
import time
import urequests
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(secrets.SSID, secrets.PASSWORD)
print(wlan.isconnected())
astronauts = urequests.get("http://api.open-notify.org/astros.json").json()
number = astronauts['number']
for i in range(number):
    print(astronauts['people'][i]['name'])`
    
    
`Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'urequests'`

Gadgetoid, 2022-07-27:
Hmm, I see why. Let me see if I can fix that.

Gadgetoid, 2022-07-27:
This build should include it, and it'll be in the next release shortly! - https://github.com/pimoroni/pimoroni-pico/actions/runs/2745701291

commit 45948040 (2022-07-27): MicroPython: Grab micropython-lib for PicoW urequests.

Fix #472
commit 3e88cd86 (2022-07-27): MicroPython: Grab micropython-lib for PicoW urequests.

Fix #472
closed 2022-07-27 by Gadgetoid