pimoroni/pimoroni-pico #484: Badger 2040 got stuck with the latest firmware v1.19.6

arturi, 2022-08-03:
I upgraded to `pimoroni-badger2040-v1.19.6-micropython.uf2` and it acts like the OS is not there — I can program the screen and LED, but the OS never comes up. Downgrading to `v1.19.5` seems to have solved the issue. Could it be that it’s the wrong file in the firmware, the one without the Badger OS?

alphanumeric007, 2022-08-04:
There are two custom uf2 files for the badger. One with badger OS and one without. ;)

Gadgetoid, 2022-08-04:
:thinking: I hope I haven't accidentally excluded the OS from *both* somehow :grimacing:

Gadgetoid, 2022-08-04:
Raised a fix for this in #485 -

You can try a test build from here: https://github.com/pimoroni/pimoroni-pico/actions/runs/2796019412

sctv, 2022-08-05:
Tried #485, it works! Thanks.

sctv, 2022-08-05:
[artefact](https://github.com/pimoroni/pimoroni-pico/suites/7663615190/artifacts/319388266)

PR #485 (2022-08-06): BadgerOS: Fix with-modules build.
Should fix #484
closed 2022-08-06 by Gadgetoid