pimoroni/pimoroni-pico #490: Issue with dithering on jpegdec

Aholland-crypto, 2022-08-09:
Hello,

While loading an image using jpegdec I'm getting a whole Buch of dots all over my image.
I am using "DISPLAY_PICO_DISPLAY_2"with a custom buffer size: display_buffer = bytearray(320*172*2)
I'm using PEN_RGB332 for the colors but I don't see an option for removing dithering.
I had it working well with PEN_RGB565 but due to memory constraints and the fact that I am no longer using a 320x240 screen, the buffer is now too small to use PEN_RGB565.

Hardware: Pimoroni Pico lipo
Library: Pico Graphics
Screen: Adafruit 1.47" 320x172 
How do I show the full image without dithering?
I am running Micro Python.

Code Snippet to load the JPEG:
    j = jpegdec.JPEG(LCD)
    # Open the JPEG file
    j.open_file("Capture.jpg")
    # Decode the JPEG
    j.decode(25, 60, 3)
    # Display the result
    LCD.update()

Here is what is available to me with jpegdec
I don't see dithering options
[image]

Gadgetoid, 2022-08-10:
That was, uh, planned:

https://github.com/pimoroni/pimoroni-pico/blob/82756a3fc99dd587ef2b77d629748e713dd4074f/micropython/modules/jpegdec/jpegdec.cpp#L129

But yes RGB332 mode should support quantized output.

Aholland-crypto, 2022-08-10:
I do have another solution I had developed loading a RAW image type. But the solution is very slow and requires the file to be converted to a RAW format. I was just hoping that maybe I missed the option for dithering and would prefer to load the image directly without dithering. Also, I would like to use the baked in libraries as much as possible.
Though the issue with speed is not really a problem for my current project since it is just a logo displayed on startup.

Gadgetoid, 2022-08-11:
Should be fixed by #492 

If you grab the appropriate build from https://github.com/pimoroni/pimoroni-pico/actions/runs/2839220919 and call decode like so: `jpeg.decode(0, 0, dither=False)` then it should posterise to the RGB332 colourspace instead.

PR #492 (2022-08-22): JPEGDEC: Add option to disable dither on RGB332.
Should fix #490 

(if it works)

Added as a new param to `decode`, use `dither=False` to disable (defaults to enabled):

```python
from picographics import PicoGraphics, DISPLAY_INKY_PACK
import jpegdec

graphics = PicoGraphics(DISPLAY_INKY_PACK)

jpeg = jpegdec.JPEG(graphics)
jpeg.open_file("128.jpeg")
jpeg.decode(0, 0, dither=False)

graphics.set_update_speed(1)
graphics.update()
```
closed 2022-08-22 by Gadgetoid