pimoroni/pimoroni-pico #494: Where is the picographics module - for use with the Micropython examples now available

ph1lj-6321, 2022-08-17:
Hi
Please provide a link to this module, so I can test the Pimoroni Display 2 on my pico
I have uploaded to the pico the latest 1.19 uf2 firmware, with Badger2040

helgibbons, 2022-08-17:
Picographics is included in the pre-built MicroPython images available on the 'Releases' page, you don't need to download the module separately. 

I don't think it's included in the Badger 2040 build though, if you're using a Pico you'll want the plain 'pimoroni-pico' one - [here's a link](https://github.com/pimoroni/pimoroni-pico/releases/download/1.19.7/pimoroni-pico-1.19.7-micropython.uf2)!

ph1lj-6321, 2022-08-17:
Thank you - I didn't release its only in the pre-release .uf2

helgibbons, 2022-08-22:
Picographics should have been included in previous releases too, I think it's just not in the Badger build (yet!)

Closing this as I think we're now sorted :)

closed 2022-08-22 by helgibbons