pimoroni/pimoroni-pico #495: ReadTimeoutError: PMS5003 Read Timeout: No response after reset

sdior501, 2022-08-21:
I keep getting this error with PMS5003 plugged to PicoEnviro+

>>> %Run -c $EDITOR_CONTENT
particle.py - Continuously print all data values.
and draw a pretty histogram on display

Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
  File "/lib/pms5003/__init__.py", line 212, in __init__
  File "/lib/pms5003/__init__.py", line 288, in reset
ReadTimeoutError: PMS5003 Read Timeout: No response after reset

What am I doing wrong?

helgibbons, 2022-08-22:
Have you checked that the cable is securely plugged in at both ends?

Might be worth unplugging the device and restarting Thonny if you haven't tried that already?

sdior501, 2022-08-22:
yes, thank you for reply. I actually checked that at least 20 times (switched ends too) as it was also my first thought. The problem is also that, it gives the same error without the sensor plugged. Sensor acts like dead, no fan, nothing. Unless you experts tell me the code/drivers are not good for the pico-enviro+ board, I will contact the vendor for a substitute.

helgibbons, 2022-08-23:
This example code is working fine for me, so I agree that it sounds like you might have a faulty sensor - if you bought it from us please drop a message to support with the order number and a link to this issue and they'll get it sorted for you!

https://shop.pimoroni.com/pages/contact-us

helgibbons, 2022-09-06:
Closing for now, but let us know if the replacement doesn't fix things!

closed 2022-09-06 by helgibbons