pimoroni/pimoroni-pico #496: Examples fail to run

FraserElectronics, 2022-08-22:
I have installed pimoroni-picow-v1.19.6-micropython.uf2 on my Pico W then copied the code from [pimoroni-pico](https://github.com/pimoroni/pimoroni-pico)/[micropython](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython)/[examples](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples)/breakout_bme280/ into Thonny. 
When I try and run the code I get RuntimeError: BreakoutBME280: breakout not found when initialising.

What an I missing?

Andy

helgibbons, 2022-08-23:
How are you connecting the breakout to your Pico?

FraserElectronics, 2022-08-23:
@helgibbons Okay, I might be missing something but I have purchased Pimoroni PIM472 and planned to connect it to the I2C or SPI port of the Pico. Do I need some additional breakout board in between the Pico and PIM472 ?

Andy

helgibbons, 2022-08-23:
Nope, you can connect it directly to the Pico's GPIO if you want! The example expects SCA on the breakout to be connected to GP4, and SCL to be connected to GP5 though, so if you're wiring it to different I2C pins you'll need to change those in your code.

FraserElectronics, 2022-08-23:
@helgibbons Okay, I think I am being dumb :-(
I think the key word in the error message is 'initialising' - I don't actually have the board connected yet, was just checking if everything else worked - so the library I suspect is actually trying to communicate with the board at this point and cannot find it. I am sure when I wire it up, all will work fine.

Andy

helgibbons, 2022-08-23:
Aha, yep - the example checks for a BME280 breakout connected to the default I2C address (0x76). If it doesn't find it will give you an error.