pimoroni/pimoroni-pico #500: Can't connect Thonny 4.0.0 using Pimoroni Pico W Enviro 1.19.7 UF2

djbloc, 2022-08-27:
Bought a Enviro Indoor with Pico W included.

Using Ubuntu 18.04 PC and Thonny 4.0.0, Thonny is unable to connect to the Pico.

When I upload the Raspberry Pi Pico W UF2 (rp2-pico-w-20220826-unstable-v1.19.1-337-gbd4e45fd6.uf2) to the device, Thonny automatically connects and MicroPython shell appears and files on the Pico are shown. 

However, if I load the 1.19.7 UF2 ([pimoroni-picow_enviro-1.19.7-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/1.19.7/pimoroni-picow_enviro-1.19.7-micropython.uf2)) Thonny does not connect (include trying all the manual ports) and no files are shown

helgibbons, 2022-12-01:
Sorry for the lack of reply here! There's a new version of the Enviro firmware at https://github.com/pimoroni/enviro/releases/tag/v0.0.9 - perhaps you could give it a go and see if you're still encountering connection issues?

ZodiusInfuser, 2023-01-20:
This issue is fixed in the release @helgibbons linked to, so I am closing this. If you experience any future issues with Enviro, please raise them over in the Enviro repo: https://github.com/pimoroni/enviro

closed 2023-01-20 by ZodiusInfuser