pimoroni/pimoroni-pico #507: Modified Launcher not loading up when reset button is pressed or when A+C pressed

realkarimsaleh, 2022-09-02:
So I have made a modified Launcher with a couple of badges I can switch between. And apparently when I press A+C all of my icons that I have added disappears.

Gadgetoid, 2022-09-02:
Hmmm I think this is because `main.py` is just `import _launcher` but it should be something like:

```python
try:
    import launcher
except ImportError
    import _launcher
```

You can probably make this tweak manually to fix your setup.

Looks like this is a bug in `boot.py` that doesn't account for a custom launcher when writing the default `main.py`: https://github.com/pimoroni/pimoroni-pico/blob/05ef0d7f037ad5050d8612d23c9829e0e6f29204/micropython/examples/badger2040/assets/boot.py#L5

realkarimsaleh, 2022-09-02:
Hey thanks for your suggestion appriicate your help. 

I have tried it and when pressing reset the E Ink display didm't refresh. 

 ```python
 try:
     import launcher
 except ImportError
     import _launcher
 ```
Honetly I don't know why it didn't work cause it looks write soo 

So I have thought of renaming the ammanded Launcher to launcherNew

and have edited `boot.py` to look something like this

 ```python
try:
    open("main.py", "r")
except OSError:
    with open("main.py", "w") as f:
        f.write("import _launcherNew")
        f.flush()
 ```

and have edited `main.py` to 
 ```python
import launcherNew
 ```

Just tested it and it works now

Gadgetoid, 2022-09-02:
Oh it *should* have been:

```
try:
    import launcher
except ImportError:
    import _launcher
```

note the missing ":". I wondered why the syntax highlighting was broken! :facepalm:

ZodiusInfuser, 2023-01-20:
Closing this issue as it appears the code offered by Gadgetoid fixed the wrong launcher loading. If it did not, please reopen.