pimoroni/pimoroni-pico #509: MSA301 example running but not getting data

AtomicFire, 2022-09-02:
Hi, running the MSA301 example the code seems to work, but I get this:
`Found MSA301. Part ID: 0x00`
then all updates read as this:
`X: 0.5002441,	Y: 0.5002441,	Z: 0.5002441,	Freefall? False,	Orientation: 0`
Obviously not giving data, even when moved.

Similarly, for the 11x7 LED matrix breakout example, the code seems to run but I see no lighting on the breakout whatsoever.

I've tested a BME688 according to the examples and that works fine, so at least some breakouts are working, just wondering if a software issue or a hardware one.

Gadgetoid, 2022-09-27:
Are these modules plugged directly into the Pico explorer base?

I might have to run some tests here and see if I can replicate.

AtomicFire, 2022-09-27:
Yes, they are, have tried different slots and no difference

helgibbons, 2022-09-27:
Are you using the correct I2C pins for Pico Explorer? With Pico Explorer Base you'll need to change the line in the example that sets up i2c to >

`i2c = PimoroniI2C(**PINS_PICO_EXPLORER)`

(Pico Explorer uses GP20 and GP21 for i2c, instead of the default GP4 and GP5).

AtomicFire, 2022-09-27:
Ah, this could be the reason, I'll check this weekend and get back to you

JeremyBYU, 2022-10-01:
I just wanted to say that I was also experiencing this issue. I had to change the pin reference and reseat a few times. Here is the working program:

```python
import time
from pimoroni_i2c import PimoroniI2C
from pimoroni import PICO_EXPLORER_I2C_PINS
from breakout_msa301 import BreakoutMSA301

i2c = PimoroniI2C(**PICO_EXPLORER_I2C_PINS)
msa = BreakoutMSA301(i2c)

part_id = msa.part_id()
print("Found MSA301. Part ID: 0x", '{:02x}'.format(part_id), sep="")

msa.enable_interrupts(BreakoutMSA301.FREEFALL | BreakoutMSA301.ORIENTATION)

while True:
    print("X:", msa.get_x_axis(), end=",\t")
    print("Y:", msa.get_y_axis(), end=",\t")
    print("Z:", msa.get_z_axis(), end=",\t")
    print("Freefall?", msa.read_interrupt(BreakoutMSA301.FREEFALL), end=",\t")
    print("Orientation:", msa.get_orientation())
    time.sleep(0.1)
```

AtomicFire, 2022-10-02:
Thanks @helgibbons and @JeremyBYU - this was the issue and Jeremy's reference code worked.

closed 2022-10-02 by AtomicFire