pimoroni/pimoroni-pico #510: ImportError: no module named 'picowireless'

slabua, 2022-09-04:
Pico W examples don't seem to work on [pimoroni-picow-1.19.7-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/1.19.7/pimoroni-picow-1.19.7-micropython.uf2)
Actually I noticed that picowireless imports successfully with [pimoroni-pico-1.19.7-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/1.19.7/pimoroni-pico-1.19.7-micropython.uf2) but the examples hang.

alphanumeric007, 2022-09-04:
I do believe that picowireless is for the Pimoroni Pico WIFI Pack, not the Pico W. The pico_wireless examples are for the pack not the Pico W.
https://forums.pimoroni.com/t/pico-w-scan-networks-looking-for-micro-python-code/20107/5

slabua, 2022-09-05:
Ah! My bad, I had totally forgotten. Thank you!

closed 2022-09-05 by slabua