pimoroni/pimoroni-pico #512: How to use micropython standard libraries?

marev3, 2022-09-14:
I want to use for example socket and ssl module from micropython, which is listed here
[http://docs.micropython.org/en/latest/library/index.html#python-standard-libraries-and-micro-libraries](url)

With
`import socket`

I get
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'socket'`

How can I get this working?

alphanumeric007, 2022-09-14:
What uf2 file are you using? 
Pico, Pico W or custom RP2040 board?

marev3, 2022-09-14:
Thank you very much.
Your answer guided me to the solution.
I have a Pico W but had uf2 file for Pico installed.

Thanks again.

helgibbons, 2022-09-20:
Sounds like we're sorted (thanks @alphanumeric007 !) - closing for now.

closed 2022-09-20 by helgibbons