pimoroni/pimoroni-pico #521: Picographics: wordwrap ignored for vector fonts

anomaly256, 2022-09-27:
Hi, I've noticed today that passing a wordwrap value though .text() is ignored when using a vector font.  Is this expected behaviour?  Documentation doesn't mention that it is.

Thanks

Gadgetoid, 2023-05-25:
Pretty much exactly what @anomaly256 said. It's a big effort for little return.

I'm actually someday hoping to replace all of our vector fonts with a polygon renderer and "Alright Fonts" - https://github.com/lowfatcode/alright-fonts

Made the first step for that here, but there's a way to go - https://github.com/pimoroni/pimoroni-pico/pull/783

fighterguard, 2023-10-23:
I've noticed something else: Line breaks don't work with vector fonts either. They only work with bitmap fonts.
I looked up and down in the docs, and there is no mention of anything related to that. So, even if you know where exactly you want a new line, it won't work. I guess vector fonts are completely useless, and should never be used.

GreenBattery, 2024-02-20:
@anomaly256 Measurement of text width appears to work correctly when it comes to centralizing text, so it seems like that aspect is already solved perhaps?

Gadgetoid, 2026-08-27:
In theory all the fonts here should still work - https://github.com/pimoroni/picovector-fonts/releases