pimoroni/pimoroni-pico #534: Backlight on Pimoroni Pico Display (135x240) Does not work

socratesv, 2022-10-07:
Hi, 

I have tried Release 19.6 and 19.8 on a Pico attached to a DS3231 RTC and a Pimoroni Pico Display (135x240) and whilst all the functions that I use work, the Backlight stays on full no matter what the setting is.

The same code on the larger 240x320 Pimoroni display works fine.

Can someone have a look please?

Many thanks

helgibbons, 2022-10-10:
That's odd - how do you have the RTC connected?

socratesv, 2022-10-13:
Thank you for looking into this.

The RTC is sandwiched between the Pico headers and the display. It uses I2C so it does not interfere with the display.

The display, RTC and Pico worked fine in 18.x using the older display code and the brightness worked fine - so it is not a hardware problem...

Hope it helps.

helgibbons, 2022-10-14:
Hmm, what pins are you using for I2C?

socratesv, 2022-10-18:
It is a piggyback pass thru. It uses pins 20 and 21 (i2c_scl = 21, i2c_sda = 20)

I guess if these are used for PWM then it will not work ( I have another SSD7789 display and I use PWM to vary the backlight brightness.)

Looking forward to your reply.

Many thanks !

socratesv, 2022-10-19:
I had a look at the Pimoroni site regarding the Pico Display and it looks like it it uses Pins 20 and 21 - is it possible to re-assign these pins ?  I will also look at the possibility of configuring the DS3231 RTC with different I2C pins if possible...

Thank you.

helgibbons, 2022-10-19:
Yep, I suspect it's sharing GP20 that's causing problems - might be easiest to rejig things so your I2C interface uses GP4 and GP5?

socratesv, 2022-10-25:
Hello Hel,

The Waveshare PICO DS3231 RTC offers alternate addresses (GP6 & GP7) but these are also use for the picodisplay LEDs.  I have now connected GP20 & GP21 to GP4 & GP5 - which are not used **just like you suggested** -  to fool the device.  I have also changed the device driver to use those GP pins and it works well - Thank you again for all your help !

helgibbons, 2022-10-26:
Awesome, glad you got it sorted! Closing for now.

closed 2022-10-26 by helgibbons