pimoroni/pimoroni-pico #535: Where can I find hte api definition for the qrcode that is used here?

MarcKarasek, 2022-10-09:
I have googled with no luck in finding micropython (or python) api docs for the qrcode library.

Gadgetoid, 2024-06-04:
I ported the QRCode library (https://github.com/nayuki/QR-Code-generator)  to MicroPython and maintain it here: https://github.com/pimoroni/QR-Code-generator/tree/micropython/edge

There are some MicroPython stubs here detailing the API: https://github.com/pimoroni/pimoroni-pico-stubs/blob/main/pimoroni_pico-stubs/qrcode/__init__.pyi

Beyond what's shown in the examples, it doesn't really do anything much since we only implement a very basic, text-only subset of the libraries features.

thirdr, 2025-01-08:
Closed as answered.

closed 2025-01-08 by thirdr