pimoroni/pimoroni-pico #548: Pico Scroll Pack - `show_text` problems

helgibbons, 2022-10-26:
The  example code that uses `show_text` to scroll text in the [MicroPython function reference](https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/pico_scroll#show_text) doesn't seem to work properly - I wonder if something in the function is overflowing somewhere?

The `scroll_text` function works fine.

```
import picoscroll
import time

# Initialise the board
picoscroll.init()

word = "Hello, world!"
l = len(word) * 6
for j in range(-17, l):
    picoscroll.show_text(word, 8, j)
    picoscroll.update()
    time.sleep(0.1)
```

https://user-images.githubusercontent.com/50950368/198073564-725f7da5-f178-425b-86e7-56af6aaecf74.mp4

Gadgetoid, 2023-03-13:
Completely refactoring this in https://github.com/pimoroni/pimoroni-pico/pull/711 so hopefully that'll fix this bug.

Should probably make it PicoGraphics compatible at some point, too...

helgibbons, 2023-03-14:
Working properly for me with #711, ta.

PicoGraphics for Pico Scroll and Pico Unicorn would be schweet!

closed 2023-03-14 by helgibbons