pimoroni/pimoroni-pico #551: Builds that have ulab preinstalled

Darkai87, 2022-10-27:
Which of the 12 builds could I download that have the ulab library already built in? Just curious because I can't see any information on which builds have it and which don't for the raspberry pi pico. Thanks.

helgibbons, 2022-11-01:
Hello! It's currently the builds for boards that have extra flash memory that include `ulab` - that's the Pimoroni Pico LiPo builds and Tiny 2040 8MB.

(`ulab` is a chunky module and I believe it was making the file sizes unworkably large for 2MB boards!)

yezzer, 2022-11-14:
@Darkai87 if it's of interested, i've loosely-documented [how to compile new Pico firmware with ulab numpy included](https://medium.com/@iestynlloyd/galactic-unicorns-and-custom-pimoroni-pico-firmware-38dd7c5913b8)

Darkai87, 2022-11-16:
@yezzer would you be able to send me your copy of your uf2 file with ulab numpy included? I'm struggling to compile this pico firmware with your instructions because I'm not the most github inclined.

yezzerfv, 2022-11-17:
> [Darkai87](/Darkai87)

DM me on twitter - same username

yezzer, 2022-11-17:
@Darkai87 dm me on twitter - same username

Gadgetoid, 2024-03-11:
I've added a note to the latest release to clarify that ulab is available in: Cosmic Unicorn, Galactic Unicorn, Stellar Unicorn, Tufty 2040, Pico LiPo 4MB and Pico LiPo 16MB.

Basically anything with > 2MB flash *or* a product I can get away with pruning some other libraries from (Cosmic, Galactic and Stellar are carefully pruned to fit ulab).

closed 2024-03-11 by Gadgetoid