pimoroni/pimoroni-pico #566: Pico Display 2.0 LED documentation

vmalisz, 2022-11-09:
hello, I can't manage to find any documentation about the led of the PIco Display 2.0. 
In the rainbow example I found the pin to use: RGBLED(6, 7, 8) then it's fairly easy from there to change the color using r,g,b values but that's it. I'd like to know how to turn it off for example or change the intensity

thank you!

vmalisz, 2022-11-09:
Looks like the pimoroni library don't have much control over the led. https://github.com/pimoroni/pimoroni-pico/blob/c5d4865609f1cb1d5bf4773d9f648885305022e6/micropython/modules_py/pimoroni.py

I ended up using this:

`from machine import Pin

led_r = Pin(6, Pin.OUT)
led_g = Pin(7, Pin.OUT)
led_b = Pin(8, Pin.OUT)

led_r.value(1)
led_g.value(1)
led_b.value(1)`

helgibbons, 2022-11-09:
The `set_rgb` helper function in `pimoroni.py` accepts a value in RGB (from 0 to 255) - so you could turn off the LED with:

```
from pimoroni import RGBLED
led = RGBLED(6, 7, 8)

led.set_rgb(0, 0, 0)
```

`led.set_rgb(127, 127, 127)` would give you half brightness.

Agree that we could use some documentation for these helper functions!

vmalisz, 2022-11-09:
omg, I should have slept on it. My brain must have crossed some wires when I asked, it makes a lot of sense :) thanks!