pimoroni/pimoroni-pico #573: Galactic Unicorn won't display JPGs correctly

yezzer, 2022-11-12:
I have partially fixed this, following @Gadgetoid's pointers.

Using the included jpegdec, an image would load in and display incorrectly:
**Incorrect**
[image]
**Correct** (Note: 8x8 pixels)
[image]

Editing jpegdec.cpp and adding the following in `int JPEGDraw(JPEGDRAW *pDraw)` resulted in the colours displaying correctly
```
else if(pDraw->iBpp == 16) {
        for(int y = 0; y < pDraw->iHeight; y++) {
            for(int x = 0; x < pDraw->iWidth; x++) {
                int i = y * pDraw->iWidth + x;
                current_graphics->set_pen(RGB(pDraw->pPixels[i]).to_rgb888());
                current_graphics->pixel({pDraw->x + x, pDraw->y + y});   
            }   
        }
```
**Fixed colours**
[image]

However, the last few columns of the image repeats.
This is not true of all sizes:
[image]

I don't do C++, so am a bit lost now. I've been able to figure out that `pDraw->iWidth` in `int JPEGDraw(JPEGDRAW *pDraw)` is incorrect though.

If anyone can make any suggestions, I can continue to try, but feel i'm unable to progress now. :)

Gadgetoid, 2022-12-13:
The auto builds here should at least now have the right pixel type - https://github.com/pimoroni/pimoroni-pico/pull/600

But I'm not sure what's going on with the width. Looks like your 11x11 pixel example is going wrong?

yezzer, 2022-12-13:
> The auto builds here should at least now have the right pixel type - #600
> 
> But I'm not sure what's going on with the width. Looks like your 11x11 pixel example is going wrong?

Yeah, some width JPGs have additional columns/rows. I haven't looked into it further as I paused dev on this project.

Gadgetoid, 2022-12-13:
I've synced JPEGDEC with the latest upstream changes to see if that's the cause, I'll have a poke.

Don't happen to have your 11x11 jpeg handy, do you?

commit 3538b6c6 (2022-12-14): JPEGDEC: Switch to iWidthUsed.

Corrects clipping width, noticable particularly on smaller jpegs and those that aren't a multiple of the MCU sized.

Fixes #573
commit 13414b50 (2022-12-14): JPEGDEC: Clip drawing to iWidthUsed.

Corrects clipping width using new iWidthUsed value, noticable particularly on smaller jpegs and those that aren't a multiple of the MCU sized.

Fixes #573
commit d977ca78 (2022-12-19): JPEGDEC: Clip drawing to iWidthUsed.

Corrects clipping width using new iWidthUsed value, noticable particularly on smaller jpegs and those that aren't a multiple of the MCU sized.

Fixes #573
closed 2022-12-16 by Gadgetoid