pimoroni/pimoroni-pico #575: Using image.py with pico inky pack

adamwest9, 2022-11-13:
I can use this to display an image o  the inky pack but I want to adjust the buttons to work with  the inky buttons but I can't  find out how to change the configuration.

helgibbons, 2022-11-15:
There's some example code for setting up and reading the buttons on Inky Pack here, does that help?

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_inky/button_test.py

adamwest9, 2022-11-15:
I wanted to use .bin files to save space. Maybe I should be asking how to display .bin files with mircopyrhon over on the inky pack section. I want to use the handy total_images count number of bin files that they used on badger image.py. I did try to reasign the pin number to make it work but that didn't work.

adamwest9, 2022-11-16:
I established a few things, one is the the buttons on the inky pack are "machine.Pin.PULL_UP" pins and not DOWN pins. so I can get the button.py for the badger 2040 to read the buttons on the inky pack but I can figure out how to adjust those pins inside of the image.py file.

adamwest9, 2022-11-17:
So now I figured out enough to know that "display.pressed(badger2040.BUTTON_A):" is always reading press, I assume it is because the inky pin is a PULL_UP and the Badger is a PULL_DOWN. If I could only figure out how to switch display.pressed to read press the other way around then everything would work (I think). If  tell it to look only  for:
if display.pressed(badger2040.BUTTON_A):
        if state["current_image"] < TOTAL_IMAGES - 1:
            state["current_image"] += 1
            changed = True
then it will flip through the .bin files until it gets to the end so it is always thinking the button is pressed. Any help would be great.