pimoroni/pimoroni-pico #582: lightsleep does not save power on pimoroni  pico uf2for 1.19.8 1.19.9 or 1.19.10 but does on picow

ssotheremail, 2022-11-19:
hi,
19/11/22

About 4 weeks since 1.19.9. pico uf2 1.19.10  appears to say its built against micropython 1.19 as against picow uf2 still appears built against 1.19.1. Put it on the pico, ran the same test and lightsleep still doesn't seem to reduce power consumption on the pico. Obviously don't know what else may not work. Previously reported via support as below but now I have a bit of time to put it on github.

via pimoroni support
20/10/22

Very rushed for time at present but noted 1.19.9 of the uf2 was out yesterday. Over 2 weeks since 1.19.8. It still appears to say its built against micropython 1.19 as against picow uf2 still saying built against 1.19.1. Impressive list of commits on github, but put it on the pico, ran the same test and (perhaps not surprisingly) lightsleep still doesn't seem to reduce power consumption on the pico. Obviously don't know what else may not work. Cant see a bug raised for this, forgive me for not doing it myself.

11/10/22
 
        If I use the current Raspberry pi uf2 it seems to be based on micropython 1.19.1 as does the picow uf2 1.19.8
        rp2-pico-20221006-unstable-v1.19.1-528-gb8982ec5f.uf2 (this is a much clearer way to version it I think).

        The Pimoroni uf2 appears to be based on 1.19.0 (which is not mentioned as a release on the MicroPython site)
        but it is referred to by pimoroni as 1.19.8
        pimoroni-pico-v1.19.8-micropython.uf2 also built on the 6th of october.

        The reason I have looked into this is power consumption in lightsleep. 
        Both uf2s support this without complaining when I do:
        machine.lightsleep(5000)

        On the Raspberry Pi uf2 or Pimoroni picow uf2 power consumption falls to 1.5 mA drain on the usb, and goes back to
        around 20mA after the sleep ends. 
        However with the Pimoroni pico uf2 it stays around 20mA all the time  The time delay provided by the sleep period seems to
        be correct in both cases.

        with pico uf2 this is on the same pico, just with different uf2s. The picow uf2 was tried on a picow.

        The reason I used the Pimoroni uf2 was because I have a Pico Display Pack 2.0 and the Raspberry Pi uf2 does not support it.

        Test program-
        
        import time
        import machine
        while True:
		time.sleep(5)
		machine.lightsleep(5000)

helgibbons, 2023-01-18:
`Lightsleep` (and a bunch of other upstream changes) should included in [v1.19.12](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.12)

closed 2023-01-18 by helgibbons

ssotheremail, 2023-01-20:
> pimoroni  pico uf2for 1.19.8 1.19.9 or 1.19.10 but does on picow (Issue
> #582)
>
> Closed #582 <https://github.com/pimoroni/pimoroni-pico/issues/582>  as
> completed.

Thanks for getting back to me with the news.

  Its brilliant that so many people put such effort into supporting 
people who want to use products like the pico by producing products like 
the pico displays etc and the software to support them.

The only other observation I would make is that perhaps it could be 
plainer which version of Micropython you are building against. I spent a 
while trying to work out what I might be doing wrong with lightsleep as 
at that point I had never used github (other than to download your uf2). 
When I spent some time on github I noted the comments about pico w using 
a different version of Micropython and, as I had one, tried it and 
realised the difference. I would hope people would not have to go to 
those lengths.

Fair enough, there does seem a dilemma to face with micropython as they 
seem to stick to a six month release interval, so it takes a while for 
changes to arrive in the release, and therefore should you build against 
release or unstable? BTW Micropython's github appears to say Jun 16, 
2022 v1.19 is its latest release which doesn't help, as clearly 1.19.1 
is about and something fairly radical has happened in Jan 23. It doesn't 
help that Micropython doesn't give a clear version statement when it 
starts. Also it seems Raspberry Pi themselves have stuck to linking to 
"unstable" versions for months, e.g. today: 
rp2-pico-20230119-unstable-v1.19.1-824-g30db33d1e.uf2, but at least 
that's clear.

BTW2 just looked more closely at your anouncement for 1.19.12:

    * Wireless boards: ***@***.*** 
<https://github.com/micropython/micropython/compare/9dfabcd6d3d080aced888e8474e921f11dc979bb...699477d12d1aa1cb57ff6cfc3f5dc5e97524dbe6>
    * Non-wireless boards: ***@***.*** 
<https://github.com/micropython/micropython/compare/v1.19...699477d12d1aa1cb57ff6cfc3f5dc5e97524dbe6> 
<https://github.com/micropython/micropython/compare/v1.19...699477d12d1aa1cb57ff6cfc3f5dc5e97524dbe6>
same build but differently labelled? internally seems to be 1.19.1

On pico 1.19.12 announces:
MicroPython 699477d on 2023-01-16; Raspberry Pi Pico with RP2040

Good news! Lightsleep does indeed work properly on the Pico with your 
1.19.12.

Anyway, I  won't take up any more of your time.

Thanks again.

Gadgetoid, 2023-01-20:
It is, as you have correctly surmised, a complete and total mess.

The release of MicroPython 1.20 (possibly with Bluetooth support) will give us a new stable release for Pico W/Pico but I suspect Bluetooth will launch with... bugs... and that customers will want us to release unstable builds to keep on top of any that are fixable in MicroPython itself.

I don't know quite how we'll move forward yet, but I do agree that having a clearly marked and current "unstable" build (perhaps generated via a separate GitHub project on a schedule) would be valuable.