pimoroni/pimoroni-pico #588: [Badger 2040] E-paper display is flickering on update with turbo speed

niutech, 2022-12-01:
I've flashed the latest version 1.19.10 of Badger 2040 firmware with BadgerOS and run the Clock demo, but the whole e-paper display is flickering on update every second, instead of just changed digits. The update speed is set to turbo. Have a look at the video below. Is it normal?

https://user-images.githubusercontent.com/384997/205068677-cf580af3-1ccd-4340-a46a-1df6a9c491d7.mp4

helgibbons, 2022-12-01:
Yep, mine looks like this too. It looks like the clock example uses the standard `update` function (i.e. refreshes the whole screen every time), I don't think it does anything fancy with `partial_update`?

niutech, 2022-12-01:
No, it doesn't use `partial_update()`. Maybe it's time to fix it?

niutech, 2022-12-02:
So everybody experience this flickering in turbo mode? Shouldn't only changing digits flicker instead of the whole screen?

niutech, 2022-12-04:
I've tested only partial refresh, but still there is a border flickering and it doesn't look good:

https://user-images.githubusercontent.com/384997/205492792-00d2fef7-3d02-4d97-b15d-0b1b46706df7.mp4

Can you fix it somehow?

Gadgetoid, 2023-03-09:
Partial update is a display controller feature, I don't believe we can change how it works at all. I suspect it's got something to do with one side of the e-paper display being energised in order to do the partial update, and that being sufficient enough to disturb the ink.

For future discussion, Badger 2040 and Badger 2040 W now live here: https://github.com/pimoroni/badger2040/

closed 2023-03-09 by Gadgetoid