pimoroni/pimoroni-pico #591: [Badger 2040] Strange lines and ghost on e-paper

niutech, 2022-12-02:
I left my Badger 2040 laying disconnected overnight with blank screen and this morning I saw strange dark lines (strokes), a border and a white ghost of an image I displayed yesterday, which should be cleared. What is causing these lines and is it normal?

[image]

Gadgetoid, 2023-03-09:
Is that the Chrome dinosaur?

In my experience e-paper is prone to weirdness like this. It pulls (or flips) ink particles in a substrate and they're not very well-behaved. This is odd, but not alarming. We might be pushing things a little far on TURBO :grimacing:

niutech, 2023-03-09:
Yes, this is the ghost from my [Dino Game](https://github.com/niutech/dino-badger2040).