pimoroni/pimoroni-pico #598: Module st7789 problems

mobigroup, 2022-12-12:
1. Module st7789 exists only in [Version 1.18.8 - Beta Bonanza](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.18.8) but it doesn't work properly.

2. Module st7789 on GitHub documented as totally different one: https://github.com/pimoroni/st7789-python
The examples like to https://github.com/pimoroni/st7789-python/blob/master/examples/round.py
```
# Create ST7789 LCD display class.
disp = ST7789.ST7789(
    port=0,
    cs=ST7789.BG_SPI_CS_FRONT,  # BG_SPI_CS_BACK or BG_SPI_CS_FRONT
    dc=9,
    backlight=19,               # 18 for back BG slot, 19 for front BG slot.
    rotation=90,
    spi_speed_hz=80 * 1000 * 1000,
    offset_left=40
)
```
do not work in MicroPython 1.18.8 and are even not available in v1.19.10.

3. The module missed in MicroPython v1.19.10 installed in Thonny:
[image]

```
MicroPython v1.19 on 2022-11-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import st7789
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'st7789'
>>> import ST7789
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'ST7789'
>>> 
```

Sure, we need to have some ability to find the actual version of Pimoroni st7789/ST7789 library (and all others too) because that's impossible to track all the GitHub changes for years and look for the ways to rebuild Pimoroni libraries pack to a consistent state and find the actual documentation. How kids would do all of that? I already asked about the issue #384 but for now the problem is the same and we even can't upgrade our old firmwares and codes due to the inconsistencies. 'picographics' module is not available always and it's not consistence between the versions the documentation is lacked too.

alphanumeric007, 2022-12-12:
st7789 is handled via Picographics. 
https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/picographics
Your posted links are to Python not Micro Python. Poke around here and you "might" find what your looking for.
https://github.com/pimoroni/pimoroni-pico/tree/main/micropython

mobigroup, 2022-12-13:
@alphanumeric007 Thanks, but there is no the main Picographics feature described (multiple displays). Does it support for now and by which way? Is it required to load st7789  directly for multiple LCDs connection or use Picographics  instead (as I remember it was "group" parameter in the early versions but it's missed in the documentation for now). Also, where to find the migration guid from the previous Picographics versions as I've asked above?

alphanumeric007, 2022-12-13:
I am currently running two Pico Display Pack V2's. It's a long read, might want to scroll to the bottom to get to what works.
https://github.com/pimoroni/pimoroni-pico/issues/299

mobigroup, 2022-12-13:
The problem is the solution based on st7789 module which is not available more. I’ve started my ticket from the point that the old code doesn’t work on new firmware but the new code or just the documentation missing the feature. While it works on my device I can’t update nether the firmware nor the code as I want to do to prepare it to be used and modified by my kids. Picographics is the new true way? Ok. But how to upgrade the old devices saving all the functions?

alphanumeric007, 2022-12-13:
I'm far from any kind of expert on this, just so you know.
My understanding is, go all new, or don't upgrade. I've been slowly going all new / Pico Graphics. When I can, I modify my old code so it works with Pico Graphics.

Gadgetoid, 2022-12-13:
Specifically which of our products are you trying to use and what are you trying to accomplish?

mobigroup, 2022-12-14:
@Gadgetoid I’ve forgot one more case to save screenshots from the display buffers. This feature is must have for oscilloscopes and other tools on RPI Pico devices. Does it work via Picographics for Pico Display Pack 2.0 (it’d be the best due to its size) and others (smaller displays are usable too if there is a problem with Pico resources)?

mobigroup, 2022-12-14:
And I found the smaller display name: Pico Display Pack. So it’s fine to use one or more Pico Display Pack (or something other maybe, like to pretty round displays) together with Pico Display Pack 2.0 for the same project connected to a single Pico.