pimoroni/pimoroni-pico #610: position_control.py typo

claypipe, 2022-12-21:
A very minor quibble.

In the Inventor 2040W micropython motor example code example position_control.py on line 38 ...

 board = Inventor2040W(motor_rear_ratio=GEAR_RATIO)
should read
board = Inventor2040W(motor_gear_ratio=GEAR_RATIO)

I note an earlier correction in the file reactive_encoder.py  with he same correction.  Thus a grep for more examples might be in order.
Cheers,

ZodiusInfuser, 2023-01-05:
Thanks for spotting this! I'm not entirely sure how I missed this. Will fix it soon.

ZodiusInfuser, 2023-01-05:
I suspect what happened is that in my testing I did `board = Inventor2040W(GEAR_RATIO)`, but added the keyword for clarity at the last moment before release.

closed 2023-01-09 by ZodiusInfuser