pimoroni/pimoroni-pico #611: Can this be imported int the arduino IDE?

SutliffeProductions, 2022-12-25:
I'm probably being super dumb. I'm trying to use the Display Pack, and i want to use it with the Arduino ide, is importing the correct libraries possible? if so, how? as there isn't any clear documentation.

helgibbons, 2023-01-18:
I don't think we've had a chance to try Pico with Arduino IDE yet - maybe someone more knowledgeable in this area can help?

ZodiusInfuser, 2023-01-20:
Hi @SutliffeProductions. The display pack uses an ST7789 LCD driver, which is pretty common. As such there are numerous guides out there for controlling it, including via Arduino. Here's one that may work for you: https://simple-circuit.com/arduino-st7789-ips-tft-display-example/

Here is the pinout for the display pack, which should help you set the appropriate pins to use the pack
[image]

closed 2023-08-08 by helgibbons