pimoroni/pimoroni-pico #617: Possible jpegdec error in Picographics Inky Pack

cjs30, 2023-01-04:
I have a problem displaying a QRcode on my Inky Pack using the jpegdec function in picographics. The jpedec puts pixels outside of the image area ([see here](https://photos.app.goo.gl/fjQXoQ5eCAbKW4am7)). I have made various QRcodes using exactly the right number of pixels to avoid aliasing and thresholded the images to white and black - and encoded with the lowest compression and no smoothing (also no progressive). jpedec must be interpreting the image dimensions incorrectly - i.e. adding to the width. Does anyone know if there is a workaround to this?

helgibbons, 2023-01-04:
Have you tried the most recent release - I think that included changes to jpegdec to better deal with small jpegs?

https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.11

cjs30, 2023-01-04:
OK, there is a work around. I have made my image 73 pixels wide - and the problem is gone! At <=72 pixels it reappears. Many thanks for the suggestion.

cjs30, 2023-01-04:
I have converted the QRcode to a binary file and have written it to the screen with the pixel function. A full 296x128 pixel(x,y) takes just over 0.5s.

helgibbons, 2023-01-04:
There's also a function for generating QR codes from your Pico if that's helpful - I don't think we have an Inky Pack example that does it but this Display Pack one should be easy to convert:

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_display/basic_qrcode.py

Gadgetoid, 2023-01-13:
I think I've now fixed what I've broken in 1.19.11 and it should be available in the (hopefully forthcoming) next release.

If you're logged into GitHub you can grab and try a pre-release build from here: https://github.com/pimoroni/pimoroni-pico/actions/runs/3911111434

This also includes some upstream fixes in MicroPython relating to wireless (particularly connecting to an open network) which might be helpful. (You probably want `
pimoroni-picow-4a88d78730904f9ca681588cff60019dc49104bb-micropython.uf2 `)

helgibbons, 2023-01-18:
A release with this fix is now available here: https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.12

Closing this one for now but do feel free to reopen it if you're still having trouble!

closed 2023-01-18 by helgibbons