pimoroni/pimoroni-pico #622: custom build: unknown type name 'time_t'

dipunm, 2023-01-08:
I'm in the process of creating a custom firmware, my plan is to make one that includes all the software I want to provide but also includes some other libraries I need.

For now, I am just trying to make a script that will build the badger2040 firmware unaltered, and then I'll figure out how I am going to include the extra libraries and copy some files in later.

This is my script:
```bash
#!/bin/bash

ROOT_DIR=$(dirname "$(which "$0")")
FIRMWARE_DIR="$ROOT_DIR/firmware"
BUILD_DIR="$FIRMWARE_DIR/build"
DEPENCENCIES_DIR="$FIRMWARE_DIR/dependencies"

MICROPYTHON_DIR="$DEPENCENCIES_DIR/micropython"
PICOSDK_DIR="$MICROPYTHON_DIR/lib/pico-sdk"
PIMORONIPICO_DIR="$DEPENCENCIES_DIR/pimoroni-pico"
PIMORONIPICO_MICROPYTHON_DIR="$PIMORONIPICO_DIR/micropython"

# Source: https://github.com/pimoroni/pimoroni-pico/blob/main/.github/workflows/micropython-badger2040.yml#L10
MICROPYTHON_VERSION=9dfabcd6d3d080aced888e8474e921f11dc979bb
PIMORONIPICO_VERSION=v1.19.10
BOARD_TYPE=PIMORONI_BADGER2040

# Install git dependencies.
if [ ! -d "$MICROPYTHON_DIR" ]; then
    pushd $DEPENCENCIES_DIR
    git clone git@github.com:micropython/micropython
    popd

    pushd $MICROPYTHON_DIR
    git checkout $MICROPYTHON_VERSION
    git submodule update --init
    popd

    pushd $PICOSDK_DIR
    git submodule update --init
    popd
fi
pushd $MICROPYTHON_DIR/mpy-cross
make
popd

if [ ! -d "$PIMORONIPICO_DIR" ]; then
    pushd $DEPENCENCIES_DIR
    git clone git@github.com:pimoroni/pimoroni-pico
    popd

    pushd $PIMORONIPICO_DIR
    git checkout $PIMORONIPICO_VERSION
    git submodule update --init
    popd
fi

pushd $MICROPYTHON_DIR/ports/rp2
$PIMORONIPICO_MICROPYTHON_DIR/_board/board-fixup.sh badger2040 $BOARD_TYPE $PIMORONIPICO_MICROPYTHON_DIR/_board

cmake -S . -B build-$BOARD_TYPE-without-badger-os \
    -DBADGER2040_NO_MODULES=1 \
    -DPICO_BUILD_DOCS=0 \
    -DUSER_C_MODULES=$PIMORONIPICO_MICROPYTHON_DIR/modules/badger2040-micropython.cmake \
    -DMICROPY_BOARD=$BOARD_TYPE \
    -DCMAKE_C_COMPILER_LAUNCHER=ccache \
    -DCMAKE_CXX_COMPILER_LAUNCHER=ccache

ccache --zero-stats || true
cmake --build build-$BOARD_TYPE-without-badger-os -j 2
ccache --show-stats || true
popd
```

heavily influenced by the github workflow, it should be doing the exact same thing.
I have the pillow python library installed
I have ccache and gcc-arm-none-eabi installed (I am running in Arch Linux though)
And of course I have all the dependencies listed here too: https://github.com/pimoroni/pico-boilerplate#preparing-your-build-environment

I am getting the following errors:

```
In file included from /[projects_dir]/firmware/dependencies/micropython/lib/mbedtls/include/mbedtls/aes.h:73,
                 from /[projects_dir]/firmware/dependencies/micropython/extmod/moducryptolib.c:64:
/[projects_dir]/firmware/dependencies/micropython/ports/rp2/mbedtls/mbedtls_config.h:109:1: error: unknown type name 'time_t'; did you mean 'size_t'?
  109 | time_t rp2_rtctime_seconds(time_t *timer);
      | ^~~~~~
      | size_t
/[projects_dir]/firmware/dependencies/micropython/ports/rp2/mbedtls/mbedtls_config.h:109:28: error: unknown type name 'time_t'; did you mean 'size_t'?
  109 | time_t rp2_rtctime_seconds(time_t *timer);
      |                            ^~~~~~
      |                            size_t
make[2]: *** [CMakeFiles/firmware.dir/build.make:2325: CMakeFiles/firmware.dir/[projects_dir]/firmware/dependencies/micropython/extmod/moducryptolib.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /[projects_dir]/firmware/dependencies/micropython/lib/mbedtls/include/mbedtls/version.h:59,
                 from /[projects_dir]/firmware/dependencies/micropython/extmod/moduhashlib.c:35:
/[projects_dir]/firmware/dependencies/micropython/ports/rp2/mbedtls/mbedtls_config.h:109:1: error: unknown type name 'time_t'; did you mean 'size_t'?
  109 | time_t rp2_rtctime_seconds(time_t *timer);
      | ^~~~~~
      | size_t
/[projects_dir]/firmware/dependencies/micropython/ports/rp2/mbedtls/mbedtls_config.h:109:28: error: unknown type name 'time_t'; did you mean 'size_t'?
  109 | time_t rp2_rtctime_seconds(time_t *timer);
      |                            ^~~~~~
      |                            size_t
make[2]: *** [CMakeFiles/firmware.dir/build.make:2353: CMakeFiles/firmware.dir/[projects_dir]/firmware/dependencies/micropython/extmod/moduhashlib.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1441: CMakeFiles/firmware.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

I have cleared ccache (`ccache -C`) this has made no difference, I am really stumped here, can anyone point me in the right direction?

dipunm, 2023-01-08:
I decided to spin up a VM with Ubuntu 22.04 and install the following packages:
`sudo apt-get install cmake ccache gcc-arm-none-eabi python3 python3-pip build-essential`

I also installed the python `pillow` package.

I ran the above script and I get a whole different set of errors :(

```
In file included from /home/dipunm/firmware/dependencies/micropython/py/mpstate.h:33,
                 from /home/dipunm/firmware/dependencies/micropython/py/runtime.h:29,
                 from /home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.h:2,
                 from /home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp:9:
/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp: In function '_PimoroniI2C_obj_t* PimoroniI2C_from_machine_i2c_or_native(mp_obj_t)':
/home/dipunm/firmware/dependencies/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:759:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  759 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_bool), assert((t) != &mp_type_bool),         \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:765:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  765 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:1093:24: note: in expansion of macro 'mp_obj_is_type'
 1093 | #define MP_OBJ_IS_TYPE mp_obj_is_type
      |                        ^~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp:15:8: note: in expansion of macro 'MP_OBJ_IS_TYPE'
   15 |     if(MP_OBJ_IS_TYPE(i2c_obj, &PimoroniI2C_type)) {
      |        ^~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:760:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  760 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_int), assert((t) != &mp_type_int),           \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:765:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  765 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:1093:24: note: in expansion of macro 'mp_obj_is_type'
 1093 | #define MP_OBJ_IS_TYPE mp_obj_is_type
      |                        ^~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp:15:8: note: in expansion of macro 'MP_OBJ_IS_TYPE'
   15 |     if(MP_OBJ_IS_TYPE(i2c_obj, &PimoroniI2C_type)) {
      |        ^~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:761:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  761 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_str), assert((t) != &mp_type_str),           \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:765:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  765 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/micropython/py/obj.h:1093:24: note: in expansion of macro 'mp_obj_is_type'
 1093 | #define MP_OBJ_IS_TYPE mp_obj_is_type
      |                        ^~~~~~~~~~~~~~
/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp:15:8: note: in expansion of macro 'MP_OBJ_IS_TYPE'
   15 |     if(MP_OBJ_IS_TYPE(i2c_obj, &PimoroniI2C_type)) {
[... cut ...]
/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp:17:15: note: in expansion of macro 'MP_OBJ_IS_TYPE'
   17 |     } else if(MP_OBJ_IS_TYPE(i2c_obj, &machine_hw_i2c_type)) {
      |               ^~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/firmware.dir/build.make:4410: CMakeFiles/firmware.dir/home/dipunm/firmware/dependencies/pimoroni-pico/micropython/modules/pimoroni_i2c/pimoroni_i2c.cpp.obj] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:1441: CMakeFiles/firmware.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

Gadgetoid, 2023-01-09:
The second set of errors you're seeing is an incompatibility between MicroPython and a whole swathe of GCC versions. This is mentioned in passing here, but otherwise quite buried - https://github.com/micropython/micropython/pull/5559

Furthermore, due to yet more weirdness with the ARM GNU Toolkit you can't build a functional MicroPython against some of our C++ code unless you use a distro-supplied version of GCC.

IMHO the best way to accomplish what you're doing at the moment is to copy our GitHub actions verbatim and let GitHub do it.

A gritty second is to run Ubuntu 20.04.5 in a VM and use the apt-supplied "gcc-arm-none-eabi".

Basically you need:

1. A version of GCC <= 9.2.1 or > 11.3
2. A stdlib that is compiled with "--disable-libstdcxx-verbose"

Failure to satisfy these will either blow up with the errors you mention above, or - if you include some of our C++ code that uses virtual functions - produce bloated MicroPython builds due to including a bunch of runtime introspection code.

The "time_t" error is odd. I've no idea what's happening there.

Building MicroPython - specially with C++ code included - is a bit of a nightmare at the moment. Good luck!

dipunm, 2023-01-09:
Hey @Gadgetoid thanks for the reply, I will look into all this. If I have to use docker or something similar to perform builds I will. I will also test using GitHub actions, although it won't be a very productive development environment.

I'll try a bunch of things and journal anything that works well here for the sake of the next developer 😉.

dipunm, 2023-01-12:
I ended up using github actions. It's a little bit of a shame that I can't have a local build environment, but it works so that's still 10x better than what I had before :)

Thanks for the help guys.

dipunm, 2023-01-12:
FYI, I will add that the only thing I managed to do that was remotely interesting before giving up was I learnt that you can add -v to the build step (not the configuration step) and it will print all the make commands that it runs and I was able to see the exact single command that failed.

It got me no where though because the error message makes sense, what doesn't make sense is why the code was missing an include to time.h or something similar. And what makes even less sense is why it works on github actions.

closed 2023-01-12 by dipunm