pimoroni/pimoroni-pico #623: mip missing from uf2 for Pico W

jon-p-taylor, 2023-01-08:
Hi guys, thanks for the great work. Is there any way of including mip in the distribution, it seems to be missing? "import mip" will fail using this uf2 but works with the latest standard micropython uf2 for the Pico W

Gadgetoid, 2023-01-09:
The groundwork is in place to switch us over to mip, but there are breaking changes in MicroPython that have to be addressed before we can build against the latest code. It's coming!

If you're logged into GitHub you should be able to grab a pre-release build here: https://github.com/pimoroni/pimoroni-pico/actions/runs/3875338726

jon-p-taylor, 2023-01-09:
That's great to hear Philip. Thanks for the update and all of the great work.

helgibbons, 2023-01-18:
Builds with `mip` should now be available here! https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.12

closed 2023-01-18 by helgibbons