pimoroni/pimoroni-pico #635: Interstate 75 w Pico example for Wii Nunchuk

antiero, 2023-01-15:
I'm trying to connect an [Adafruit Wii Nunchuck Breakout Adapter](https://shop.pimoroni.com/products/adafruit-wii-nunchuck-breakout-adapter-qwiic-stemma-qt?variant=32348080963667) to the [Interstate 75 W board ](https://shop.pimoroni.com/products/interstate-75-w?variant=40453881299027) , but really struggling to find any docs or examples for this.. I'd ideally like a MicroPython or CircuitPython example if possible - any help much appreciated!

[image]

ZodiusInfuser, 2023-01-20:
Hi, perhaps you have come cross this already, but here is some Micropython code I found that could work: https://github.com/kfricke/micropython-nunchuck

Here is the Circuitpython code for that adafruit Nunchuk adapter, which could be useful for converting to Micropython: https://github.com/adafruit/Adafruit_CircuitPython_Nunchuk/blob/main/adafruit_nunchuk.py

antiero, 2023-01-26:
Thanks @ZodiusInfuser , I got this working in the end! Will tidy up the code and share an example soon.

ZodiusInfuser, 2023-01-26:
That's great to hear, and see! Thanks for sharing

closed 2023-01-30 by ZodiusInfuser