pimoroni/pimoroni-pico #644: Plasma 2040 with sk6812 rgbw LED crashes

Gunsi1973, 2023-01-20:
This chips shoudl be supported (according to Pimoronis website). It's not entirely sure, what type has to be used, I tried both. From web reasearch I found out, that sk6812 should be used with WS2812 (Neopixel) but if I do that, I get an unresponsive Plasma 2040. If I use "plasma.APA102" the device is responsive and I can get some LEDs to work but the behaviour is erratic. Sk6812 is missing the CLK anyway. Maybe its the RGBW (White) part that is not supported or something else is wrong. Anyone with similar experiences?

crashes while constructing the object led_strip

`import plasma
from plasma import plasma2040
import time
led_strip = plasma.WS2812(30, 0, 0, plasma2040.DAT)`

helgibbons, 2023-01-20:
Hey hey - are you using v1.19.12 MicroPython? This version contained a [Plasma bug](https://github.com/pimoroni/pimoroni-pico/issues/637) that was causing everything to fall over - it will be fixed in the next release but in the meantime you could try [v1.19.11](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.11)?

helgibbons, 2023-01-20:
I've added a note to the v1.19.12 release notes to make this a bit more visible!

helgibbons, 2023-02-16:
This bug should be fixed in the [most recent release](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.14) so closing this one for now - let us know if you're still having issues!

closed 2023-02-16 by helgibbons