pimoroni/pimoroni-pico #645: Enhancement: library-only // Micropython

MaxMyzer, 2023-01-20:
Hi, 
I thought the inky display was interesting, but unfortunately, I have a pico that also needs special drivers for it, so I can't just upload this firmware. Is there any possibility in the future that instead of custom firmware, they could just be libraries? Or could parts that might be required to be in firmware get a PR into the main micropython?

ZodiusInfuser, 2023-01-21:
Hi. It is very unlikely that we will offer a micropython driver for the inky display, due to how integrated all the displays are under our PicoGraphics library. Is there any particular reason you cannot use our version of micropython?

You could always find (or write) a micropython for the inky display. It is the UC8151. I know there are drivers available for CircuitPython for example, as I have used them. so perhaps that code can be ported?

roaldarbol, 2023-02-14:
Sorry if I've misunderstood something in advance. 
I often would want to use the breakouts in conjunction with other projects; I use [Belay](https://belay.readthedocs.io/en/latest/index.html) which is amazing for controlling stuff directly from your PC. Belay has a package manager, where you can specify individual Micropython packages in `pyproject.toml` (bringing in the same unifying principles that a lot of Python packaging tools, e.g. Poetry, does). However, I can't seem to find the standalone Python libraries/files, just the C, Cmake, etc. files. Are they somewhere to be found, or is there a reason they are not? 
Say I want to use the `BH17455`, and want to point to the `.py` file - is that possible?

MaxMyzer, 2023-02-15:
> Hi. It is very unlikely that we will offer a micropython driver for the inky display, due to how integrated all the displays are under our PicoGraphics library. Is there any particular reason you [...]

Thats too bad. Luckily I was able to find another product that doesn't need a driver, and am using that.

closed 2023-02-15 by MaxMyzer

roaldarbol, 2023-02-27:
@ZodiusInfuser Ah, that's unfortunate. Would it be possible to compile Python from the C code somehow? It would immensely helpful, as all your work could be used with package managers, such as [Belay](https://belay.readthedocs.io/en/latest/Package%20Manager.html) or `mip`, but without the Python packages, that's not possible. I'm not particularly interested in the `BH1745` per se, but more in the general process of making all the amazing Micropython code available to the widest possible audience - it's a pity that the libraries can't be used outside a Pimoroni installation. I think thinking about how to integrate with package managers more seriously together with the great people at Micropython and elsewhere could unlock some great potential (see e.g. [this discussion](https://github.com/micropython/micropython-lib/issues/578)).