pimoroni/pimoroni-pico #652: How to get the Day from RV3028 RTC in Micro python?

alphanumeric007, 2023-01-27:
Playing around with the Micro Python demo for the RV3028 I can get the Month, and Weekday, but can't get the Day of the month?
`
from pimoroni_i2c import PimoroniI2C
from breakout_rtc import BreakoutRTC
import time

PINS_BREAKOUT_GARDEN = {"sda": 4, "scl": 5}  # i2c pins 4, 5 for Breakout Garden
PINS_PICO_EXPLORER = {"sda": 20, "scl": 21}  # Default i2c pins for Pico Explorer

i2c = PimoroniI2C(**PINS_BREAKOUT_GARDEN)
rtc = BreakoutRTC(i2c)

if rtc.is_12_hour():
    rtc.set_24_hour()

rtc.enable_periodic_update_interrupt(True)

while True:
    if rtc.read_periodic_update_interrupt_flag():
        rtc.clear_periodic_update_interrupt_flag()

        if rtc.update_time():
            rtc_weekday = rtc.get_weekday()
            rtc_month = rtc.get_month()
            rtc_date = rtc.string_date()
            rtc_time = rtc.string_time()
            print("Date: ", rtc_date, ", Time: ", rtc_time, ", Month: ", rtc_month, ", Weekday: ", rtc_weekday, sep="")

    time.sleep(0.1)
Adding
`rtc_day = rtc.get_day()`
gets me
AttributeError: 'BreakoutRTC' object has no attribute 'get_day'

@Gadgetoid

ZodiusInfuser, 2023-01-27:
@alphanumeric007 Give `get_date()` a go. That looks like it returns a value between 1 and 31.

alphanumeric007, 2023-01-27:
OK thanks a bunch, that worked. =)
```
date = rtc.get_date()
display.text(f"{date:02}", 160, 13, scale=3)
```
got me a 27
I missed it because `rtc_date = rtc.string_date()` gets me 27/02/2023. I missed the 'rtc.string_date' versus 'rtc.get_date' and was fixated on looking for "day" in the library code.