pimoroni/pimoroni-pico #656: Interstate 75: HUB-75E panel has blue and green swapped

PimoroniMatt, 2023-01-30:
Update to allow for panels with different arrangement of colours, so RGB/RBG/Whatever can be specified and colours appear correctly.

ZodiusInfuser, 2023-01-31:
Raised a pull request with a patch for this: https://github.com/pimoroni/pimoroni-pico/pull/660

helgibbons, 2023-02-16:
HUB75 colour order is swappable in the [most recent release](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.19.14). Here's a quick example!

```
from interstate75 import Interstate75

i75 = Interstate75(display=Interstate75.DISPLAY_INTERSTATE75_64X64, color_order=Interstate75.COLOR_ORDER_RBG)
graphics = i75.display

BLACK = graphics.create_pen(0, 0, 0)
RED = graphics.create_pen(255, 0, 0)
GREEN = graphics.create_pen(0, 255, 0)
BLUE = graphics.create_pen(0, 0, 255)

graphics.set_pen(BLACK)
graphics.clear()
graphics.set_font("bitmap8")

while True:
    # Clear display
    graphics.set_pen(BLACK)
    graphics.clear()
    graphics.set_pen(RED)
    graphics.text("red", 0, 0)
    graphics.set_pen(GREEN)
    graphics.text("green", 0, 20)
    graphics.set_pen(BLUE)
    graphics.text("blue", 0, 40)
    i75.update(graphics)
 ```

closed 2023-02-16 by helgibbons