pimoroni/pimoroni-pico #657: PicoW Plasma: unable to estabilish a WiFi connection

maiorfi, 2023-01-30:
I'm working with 1.19.11 version, because of the bug that prevents plasma boards to properly boot with 1.19.12.

I cannot manage to connect to any of 3 different (2.4GHz) AP running provided samples, while, flashing standard MicroPython firmware (both latest nightly, i.e. rp2-pico-20230128-unstable-v1.19.1-841-g3446d440f.uf2, and latest stable, rp2-pico-20220618-v1.19.1.uf2), I succeed in connecting to any of my APs (and get a IP address via DHCP).

Any suggestion?

ZodiusInfuser, 2023-01-31:
Hi, this is very odd. Are you running the exact same code on all the micropython builds you tried?

maiorfi, 2023-02-02:
Well, actually this code snippet works with both firmwares:

```python
import network
import time

from wifi_config import *
 
def do_connect(ssid=wifi_config['ssid'],psk=wifi_config['password']):
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(ssid, psk)
 
    # Wait for connect or fail
    wait = 30
    while wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        wait -= 1
        print('waiting for connection...')
        time.sleep(1)
 
    # Handle connection error
    if wlan.status() != 3:
        raise RuntimeError('wifi connection failed')
    else:
        print('connected')
        ip=wlan.ifconfig()[0]
        print('network config: ', ip)
        return ip
```

So I retried, with pimoroni firmware, running one of pimoroni picow plasma kit samples, weather.py, on a plain PicoW stick and it successfully connects to wifi access point and acquires IP address.

I will be able to test same sample (i.e. weather) on a plasma w 2040 stick later today.

I will keep you in touch.

Thanks.

maiorfi, 2023-02-04:
Ok, just cleared a little bit the issue scenario.

On SAME BOARD, I have following code snippet working properly with micropython-firmware-pico-w-290622.uf2, while not working with pimoroni-picow-v1.19.11-micropython.uf2.

```python
import network
import time

from wifi_config_json import *
 
def do_connect(ssid=wifi_config['ssid'],psk=wifi_config['password']):
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(ssid, psk)
 
    # Wait for connect or fail
    wait = 30
    while wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        wait -= 1
        print('waiting for connection...')
        time.sleep(1)
 
    # Handle connection error
    if wlan.status() != 3:
        raise RuntimeError('wifi connection failed')
    else:
        print('connected')
        ip=wlan.ifconfig()[0]
        print('network config: ', ip)
        return ip
    
# main
do_connect()
```

The actual error I get is:

```
>>> %Run -c $EDITOR_CONTENT
waiting for connection...
Traceback (most recent call last):
  File "<stdin>", line 30, in <module>
  File "<stdin>", line 22, in do_connect
RuntimeError: wifi connection failed
>>> 
```

Is there anything I am missing?

ZodiusInfuser, 2023-02-04:
Not that I can see. Of course the exception you're seeing is the one you added in, caused by `wlan.status() != 3`. I am not familiar with what 3 means in this context. I would assume from this list  (https://docs.micropython.org/en/latest/library/network.WLAN.html) that it's STAT_NO_AP_FOUND.

This could either be genuine, in which case trying your tests nearer your AP may help, or it could be a result of the WiFI chip getting into a locked-up state. Perhaps include an SSID scan at the start?

To my knowledge the Pico W's WiFi does not reset as you would expect when you reset your pico or reflash, so the only way to be sure there's no lingering state would be to disconnect it from power after reflashes, and wait a few minutes for your AP to clear any blocked timeout it may have imposed (i've personally had this with my testing).

If you try all that and issues remains exclusively with our build, then I am unsure how we can begin to diagnose that.

maiorfi, 2023-02-06:
Concerning STAT_NO_AP_FOUND constant, where could I find MicroPython library sources?

ZodiusInfuser, 2023-02-06:
I searched google for "micropython network" and looked through the links under the docs.micropython.org domain: https://docs.micropython.org/en/latest/library/network.WLAN.html

maiorfi, 2023-02-06:
Yes, but do you know how to get to python file that defines such constants?

ZodiusInfuser, 2023-02-06:
This part of the micropython code suggests they may be on the `network` module:
https://github.com/micropython/micropython/blob/9e9176467179bcb477b92f32033e62fcde14ea41/ports/rp2/mpconfigport.h#L163-L169