pimoroni/pimoroni-pico #662: 'urllib' for pimoroni-pico/micropython/examples/pico_inky

quirozity, 2023-02-02:
Are there any common examples for urlib that can be used for the picow and e ink display?
running the picow examples
ImportError: no module named 'urllib'

helgibbons, 2023-02-02:
Examples for Pico Inky Pack are here - https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/pico_inky

You should be able to install `micropython-urllib.urequest` via Thonny's 'Manage Packages' or there's a pre-compiled version here that you can copy into the `lib` file on your Pico: https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/examples/common/lib

helgibbons, 2023-02-15:
Closing this one for now but let us know if you're still having trouble!

closed 2023-02-15 by helgibbons