pimoroni/pimoroni-pico #674: Interstate 75 support different refresh

iMartyn, 2023-02-20:
I have a 64x32 panel (interestingly on the back it says P4-256*128-2121-A5) and it repeats rows and doesn't allow for full control of the panel with the interstate 75 W board.

The following code :
```python
import hub75

WIDTH = 64
HEIGHT = 32

matrix = hub75.Hub75(WIDTH, HEIGHT, stb_invert=False)

matrix.start()
matrix.clear()

for x in range(WIDTH):
    matrix.set_pixel(x,0,255,0,0)
for x in range(WIDTH):
    matrix.set_pixel(x,2,0,255,0)
for x in range(WIDTH):
    matrix.set_pixel(x,8,0,0,255)
for x in range(WIDTH):
    matrix.set_pixel(x,10,0,255,0)
for x in range(WIDTH):
    matrix.set_pixel(x,16,255,0,255)
for x in range(WIDTH):
    matrix.set_pixel(x,18,0,255,255)
for x in range(WIDTH):
    matrix.set_pixel(x,24,128,128,255)
for x in range(WIDTH):
    matrix.set_pixel(x,26,128,255,128)
```
produces this output :

[image]

As you can see, the panel is reacting correctly to the colours, but instead of a 64*32 LCD I have a weirdly interleaved 64*8 board!

As I understand it, this could be the difference between a refresh rate of different boards, so it would be really nice to let us set that and then we can figure out a mapping that works.

helgibbons, 2023-02-21:
Hello! Is that a panel you've [sourced from us](https://shop.pimoroni.com/products/rgb-led-matrix-panel?variant=42312764298), or from elsewhere?

Gadgetoid, 2023-02-21:
This has come up before in #233 and #249 and the same problems apply here- a mixture of a desperate shortage of time, energy and hardware to test with.

The pixels are arranged internally like so:

https://github.com/pimoroni/pimoroni-pico/blob/f06d7c2e83debff09d5ac69969864e7e76dff95e/drivers/hub75/hub75.cpp#L47-L58

With an interleaved pattern reflecting the fact the panel is split into two logical sections- top and bottom- which are driven by shift register chains that handle WIDTH * 2 pixels for the top/bottom segments of the display.

Pretty much the best you can do right now if you're using MicroPython is fork this repo, delete the remapping code, let GitHub Actions spit you out a build and give it a go. Rinse & repeat.

iMartyn, 2023-02-21:
@helgibbons it is an AliExpress panel I had from before the good ship started stocking them.

Thanks @Gadgetoid that feels like it will have an overly high cycle time for me too but at least I can start with a zero mapping and see if I can write to all the pixels without it.

iMartyn, 2023-02-21:
That being said, I still don't think this will make a difference because I don't think it's the mapping that's the problem but the refresh rate.  Look carefully and you'll see I'm setting 64 pixels of red and getting 64*4 pixels of red.

thirdr, 2025-01-08:
Closing this to clear our decks for the new year, if it's still troubling you please re-open! :)

closed 2025-01-08 by thirdr