pimoroni/pimoroni-pico #681: PicoGraphics feature request: disable dithering for Inky Frame

dvankley, 2023-02-22:
[Support forum thread](https://forums.pimoroni.com/t/picographics-disable-dithering-for-inky-frame/21533)

The gist is that I'm not crazy about the built-in PicoGraphics dithering for Inky Frame and would like the option to be able to provide my own. I have had success doing so with Inky Impression and would like to apply the same technique to Inky Frame.

I see [something similar](https://github.com/pimoroni/pimoroni-pico/pull/492) has been provided for RGB332, I'd basically just like the same thing for the Inky.

As part of this, it would be useful to have specific guidance on what (7 or 8) color codes to use as the palette.

Thanks!

Gadgetoid, 2023-02-22:
Disabling dither for 3bit/4bit palette pen types seems doable- I've tested it in C++ and am porting to MicroPython. The results as - as you'd imagine - pretty horrible.

You would still need to pre-dither your JPEG images and hope the compression doesn't mess it up. At that point you might as well just prepare a raw bitmap style file and push that right into PicoGraphic's buffer.

Our dithering code is currently a bit of a mess and copied more or less verbatim across 8bit, 4bit and 3bit pen types. It would be quite tricky to make it user-customizable without some significant refactoring and that's not likely to happen in a hurry.

dvankley, 2023-02-22:
@Gadgetoid thanks for the response.

>You would still need to pre-dither your JPEG images and hope the compression doesn't mess it up.

Pre-dithering is fine, that's my intended workflow. I wasn't considering the impact of compression, though, that's a good point.

>At that point you might as well just prepare a raw bitmap style file and push that right into PicoGraphic's buffer.

I'm down for that. I can alter my image prep flow to change the format as needed, which I guess would be a 3 bits-per-pixel bitmap? Any guidance you can provide on how to handle the PicoGraphics side?

Gadgetoid, 2023-02-22:
> I'm down for that. I can alter my image prep flow to change the format as needed, which I guess would be a 3 bits-per-pixel bitmap? Any guidance you can provide on how to handle the PicoGraphics side?

On Inky 5.7" you can just copy data to the buffer using, if I'm not mistaken, `open("filename.bin").readinto(picographics)`

It uses three separate bit planes to store colour information- packing the 600x448 display into 100,800 bytes containing three blocks of 33,600bytes for each bit of an individual colour value.

So `0b111` at pixel 0,0 will set the first, 38,400th and 76,800th byte to `0b10000000`, proof:

```python
from picographics import PicoGraphics, DISPLAY_INKY_FRAME
graphics = PicoGraphics(DISPLAY_INKY_FRAME)

w, h = graphics.get_bounds()

def show_pixel(x, y):
    offset = (w * h) // 8
    pixel = (x // 8) + (y * w // 8)
    buf_a = memoryview(graphics)[pixel]
    buf_b = memoryview(graphics)[pixel + offset]
    buf_c = memoryview(graphics)[pixel + offset + offset]
    print(f"Plane 1: {buf_a}")
    print(f"Plane 2: {buf_b}")
    print(f"Plane 3: {buf_c}")

print("\nEmpty")
show_pixel(0, 0)

print("\n7 at 0, 0")
graphics.set_pen(0b111)
graphics.pixel(0, 0)
show_pixel(0, 0)

print("\n7 at 1, 0")
graphics.set_pen(0b111)
graphics.pixel(1, 0)
show_pixel(1, 0)

print("\n4 at 10, 10")
graphics.set_pen(0b100)
graphics.pixel(10, 10)
show_pixel(10, 10)
```

And the `set_pixel` method can be reimplemented in MicroPython, totally bypassing PicoGraphics and drawing into the framebuffer like so:

```python
from picographics import PicoGraphics, DISPLAY_INKY_FRAME
graphics = PicoGraphics(DISPLAY_INKY_FRAME)

w, h = graphics.get_bounds()

def show_pixel(x, y):
    offset = (w * h) // 8
    pixel = (x // 8) + (y * w // 8)
    buf_a = memoryview(graphics)[pixel]
    buf_b = memoryview(graphics)[pixel + offset]
    buf_c = memoryview(graphics)[pixel + offset + offset]
    print(f"Plane 1: {buf_a}")
    print(f"Plane 2: {buf_b}")
    print(f"Plane 3: {buf_c}")

def put_pixel(x, y, c):
    offset = (w * h) // 8
    pixel = (x // 8) + (y * w // 8)
    
    bit_offset = 7 - (x & 0b111)

    buf_a = pixel
    buf_b = pixel + offset
    buf_c = pixel + offset + offset
    
    cA = (c & 0b100) >> 2
    cB = (c & 0b010) >> 1
    cC = (c & 0b001)
    
    # Clear bits
    memoryview(graphics)[buf_a] &= ~(1 << bit_offset)
    memoryview(graphics)[buf_b] &= ~(1 << bit_offset)
    memoryview(graphics)[buf_c] &= ~(1 << bit_offset)
    
    # Set bits
    memoryview(graphics)[buf_a] |= (cA << bit_offset)
    memoryview(graphics)[buf_b] |= (cB << bit_offset)
    memoryview(graphics)[buf_c] |= (cC << bit_offset)

print("\nEmpty")
show_pixel(0, 0)

print("\n7 at 0, 0")
#graphics.set_pen(0b111)
#graphics.pixel(0, 0)
put_pixel(0, 0, 0b111)
show_pixel(0, 0)

print("\n7 at 1, 0")
#graphics.set_pen(0b111)
#graphics.pixel(1, 0)
put_pixel(1, 0, 0b111)
show_pixel(1, 0)

print("\n4 at 10, 10")
#graphics.set_pen(0b100)
#graphics.pixel(10, 10)
put_pixel(10, 10, 0b100)
show_pixel(10, 10)
```

Just swap `memoryview(graphics)` for your own buffer of size `600 * 448 / 8 * 3` and this should help you convert images.

:warning: On Inky 7.3" (coming soon) this will just hardfault because its buffer is in PSRAM. I should probably guard against that :grimacing: or find some way to abstract it.

Gadgetoid, 2023-02-22:
Dither disable added as of https://github.com/pimoroni/pimoroni-pico/pull/682

It will be in the next release, or you can grab it from Actions when it's done cooking- the .uf2 file will appear here: https://github.com/pimoroni/pimoroni-pico/actions/runs/4244929564

Pretty nice results on the right type of image-

[image]

Gadgetoid, 2023-02-22:
Tried a pre-process using https://ditherit.com/, Atkinson dithering against the palette `[{"hex":"#000000"},{"hex":"#ffffff"},{"hex":"#00ff00"},{"hex":"#0000ff"},{"hex":"#ff0000"},{"hex":"#ffff00"},{"hex":"#ff8000"},{"hex":"#dcb4c8"}]` (there's an import option) and got a reasonable result. The JPEG is HUGE though.

[image]

Gadgetoid, 2023-02-22:
Sorry, to read a .bin should probably use "[b]inary" mode:

```python
from picographics import PicoGraphics, DISPLAY_INKY_FRAME
graphics = PicoGraphics(DISPLAY_INKY_FRAME)

open("dither_it_place-kitten-red-cat.inky.bin", "rb").readinto(graphics)

graphics.update()
```

And to convert a source image:

```python
from PIL import Image
import sys
import pathlib

file = pathlib.Path(sys.argv[1])

# Image to hold the quantize palette
pal_img = Image.new("P", (1, 1))

# Image *should* already be using just these colours
pal_img.putpalette((
    0, 0, 0,
    255, 255, 255,
    0, 255, 0,
    0, 0, 255,
    255, 0, 0,
    255, 255, 0,
    255, 128, 0,
    220, 180, 200
))

# Open the source image and quantize it to our palette
i = Image.open(file).convert("RGB").quantize(palette=pal_img)

WIDTH, HEIGHT = 600, 448

i = i.crop((0, 0, WIDTH, HEIGHT))

buf = bytearray(WIDTH * HEIGHT // 8 * 3)

def put_pixel(x, y, c):
    offset = (WIDTH * HEIGHT) // 8
    pixel = (x // 8) + (y * WIDTH // 8)
    
    bit_offset = 7 - (x & 0b111)
    
    b = (c & 0b100) >> 2

    buf[pixel] &= ~(1 << bit_offset)
    buf[pixel] |= (b << bit_offset)

    pixel += offset

    b = (c & 0b010) >> 1
    buf[pixel] &= ~(1 << bit_offset)
    buf[pixel] |= (b << bit_offset)

    pixel += offset
    
    b = (c & 0b001)
    buf[pixel] &= ~(1 << bit_offset)
    buf[pixel] |= (b << bit_offset)

# Copy the source image pixels into our buffer
for x in range(WIDTH):
    for y in range(HEIGHT):
        put_pixel(x, y, i.getpixel((x, y)))

# Write our bin file
with open(file.with_suffix(".inky.bin"), "wb+") as f:
    f.write(buf)
```

GLHF!

dvankley, 2023-03-04:
Suggested approach works great for me. Thanks for the help!

closed 2023-03-04 by dvankley

qpincon, 2023-07-30:
How would you advise to do the same thing for the 7.3'' display? Just by calling `pixel(x,y)` (although described as slow in the documentation), or is there any way to set the PSRAM directly?

dvankley, 2023-07-31:
@qpincon I opened an issue for that very functionality: https://github.com/pimoroni/pimoroni-pico/issues/745
I'm not sure of the current status.

Redsandro, 2024-02-08:
@Gadgetoid thank you for the binary conversion code. This is really helpful!

Is there any kind of existing indexed image format that is easy and lightweight enough for even the Inky Frame 5.7 to decode?

I'm sending images to my Inky Frame over MQTT. Now they are <10 kb jpg files so obviously the dithering is terrible and I found this topic. Mainly I'm trying to solve two problems:

1. I would like the binary messages to be smaller so my MQTT network is not congested.
2. Preferably the messages are in an existing format so I can re-use them.

Redsandro, 2024-02-11:
How do you force the palette to be in the correct order?

Both `PIL.image` and `wand.image` provide `remap()` and `quantize()` respectively, but I keep getting the colors mixed up. I get the correct image shape, but a scrambled palette. The order is not consistent with different images, so I can't simply create a LUT.

I can write the image to PNG and load it with [pngdec](https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.20.4) and it's beautiful. But once I do the framebuffer trick (because no one has 48k free memory), the colors are scrambled.

It's starting to look like the input image has influence over what color comes first in the final quantized palette, and it's different from the order defined in `pal_img.putpalette()`.

Redsandro, 2024-02-12:
I figured out my mistake. Most Pillow operations return a new Pillow object. Doing a lot of `im = im.operation()` made me do `im = im.putpalette()` which removes `im`, effectively causing `.quantize(palette=None)` resulting in a default palette.

It's working properly now. I've wasted a lot of time but I ended up with cleaner code. This function takes a binary image string (from MQTT) and converts it to a frame buffer:

```python
def crop_and_dither(jpg_bin):
    def put_pixel(buf:bytearray, x:int, y:int, c):
        offset      = (WIDTH * HEIGHT) // 8
        pixel       = (x // 8) + (y * WIDTH // 8)
        bit_offset  = 7 - (x & 0b111)

        for i in range(3):
            b = (c >> (2 - i)) & 1
            buf[pixel + i * offset] &= ~(1 << bit_offset)
            buf[pixel + i * offset] |= (b << bit_offset)

    with Pillow.open(BytesIO(jpg_bin)) as im:
            framebuffer = bytearray(WIDTH * HEIGHT // 8 * 3)
            pimg        = Pillow.new("P", (16, 16))
            pimg.putpalette(inky.PAL + (0, 0, 0) * 8 * 31, rawmode='RGB')
            im          = im.convert(mode='RGB').quantize(colors=7, palette=pimg, dither=1)

            for x in range(WIDTH):
                for y in range(HEIGHT):
                    try:
                        put_pixel(framebuffer, x, y, im.getpixel((x, y)))
                    except IndexError:
                        pass

            return framebuffer
```