pimoroni/pimoroni-pico #689: Localhost / 127.0.0.1 Badger2040w

North101, 2023-02-24:
Does localhost / 127.0.0.1 work on the picow?

I can run my server code on one picow and have another picow connect to it, but running the same server code in a thread and trying to connect to localhost / 127.0.0.1 doesn't work

`socket.getaddrinfo('localhost', 1246)` hangs
`socket.connect(('127.0.0.1', 1246))` errors with `EHOSTUNREACH`

Is there anything I'm missing?

Gadgetoid, 2023-02-27:
I would imagine not, the Pico isn't really meaningfully capable of running multiple simultaneous processes so even if there was local loopback support your request would block your response and vice versa.

What are you trying to accomplish?

North101, 2023-02-27:
Ok, that makes sense. I have 4 pico w's all talking to each other over wifi and one of them was going to act as a server and client, while the other were clients. For simplicity sake, I was hoping to reuse the client code and just connect to localhost

closed 2023-02-27 by North101

Gadgetoid, 2023-02-27:
Ah, I getcha. Should be able to abstract the core of the server/client enough that they can just use a direct function call for inter-process communication.