pimoroni/pimoroni-pico #690: Pico Display How to close RGB_LED

Hsgngr, 2023-02-25:
I see that previously there was a function called display.set_led(0,0,0) to to change the LED however now there is not such a function. How someone can close those built-in rgb_led on pico display ?

alphanumeric007, 2023-02-25:
Still works for me? providing you set it up correctly in the first place.
`
from pimoroni import RGBLED
led = RGBLED(6, 7, 8)
led.set_rgb(0, 0, 0) 

`

closed 2023-02-25 by Hsgngr

alphanumeric007, 2023-02-25:
It's listed in some of the examples.
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pico_display/rainbow.py
And the pinout is at the bottom of this page.
https://shop.pimoroni.com/products/pico-display-pack

Hsgngr, 2023-02-26:
I saw the given pinout but it says 9 10 11 instead of 6 7 8. For the example you are right.

alphanumeric007, 2023-02-26:
The pinout is listing physical pin numbers, while the code deals with the GP numbers.
GP 6 is physical pin 9
GP 7 is physical pins 10
GP 8 is physical pin 11
https://pico.pinout.xyz/