pimoroni/pimoroni-pico #691: Interstate 75, no support for panels stacked vertically?

alphanumeric007, 2023-02-25:
I have two 64 wide by 32 high panels setup for a 64 x 64 LED Matrix. Currently my only option is for a 128 X 32 array setup.
`from interstate75 import Interstate75, DISPLAY_INTERSTATE75_128X32` 
Trying to use 64X64 will not work, pixel mapping is all messed up. 

I want to scroll text on my 64 X 64 array. I can scroll text but it ends up being something like this:
my scrolling text
gets chopped.....

I have looked at any info I can find, but haven't found a way around this.  
https://learn.pimoroni.com/article/getting-started-with-interstate-75
https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/hub75#getting-started

If anybody has so insight on this I would greatly appreciate it.

Gadgetoid, 2023-02-25:
We abandoned our dreams of arbitrary panel configurations after grappling with it for some time. Our very opinionated graphics API doesn't really lend itself well to passing all this nonsense through.

We probably need to refactor Interstate 75 to work in the same way Cosmic and Galactic do- having a separate "library" to manage the display configuration, and using PicoGraphics just as a framebuffer.

This is another spin on #249, pretty much, we need robust and fast pixel mapping both to handle weird displays and custom configurations. Unfortunately not something we have the time for at the moment :grimacing:

alphanumeric007, 2023-02-26:
I had a feeling this was one of those, sounds so easy to do, but is anything but, situations. Had to ask though. I'll likely just go to a single 64 x 64 panel. That will get me readable scrolling text. I should have waited for the single panel with the wide pixel spacing I wanted to be in stock. 
If I get really board I'll have another go at Adafruits Display IO. For me that seems to be the other end of the extreme. Very feature rich, but very complicated to use. 

Thanks for the quick response.

closed 2023-02-26 by alphanumeric007

alphanumeric007, 2023-02-26:
The Galactic and Cosmic examples are what I have been going too as a start off point. I got scrolling text working on my i75. It's split level though because of my layout.
Space is big. Really bi
g. You just won’t beli
https://forums.pimoroni.com/t/interstate-75-with-multiple-hub-75-led-matrix-panels/21397/18
Also got the 80's Super Computer to run, sort of. 64 x 32 works but 128 x 32 doesn't work. It ends up just duplicating on the two displays, and doesn't look too bad. 
https://forums.pimoroni.com/t/want-to-run-80s-super-computer-example-on-interstate-75/21557/3