pimoroni/pimoroni-pico #698: jpegdec on tufty 2040 does not work from SD, only flash

quarterturn, 2023-02-27:
I have a Tufty 2040 which has been adapted to use an microsd adapter. 

I am using the sdcard library from here:
https://github.com/OneMadGypsy/pico-sd-card

It is working. I am able to read out a text file which around the size of a jpeg file (11K).

code used for testing:
`
import machine
import sdcard
import uos

sd = sdcard.SDCard(0, 6, 7, 4, 5, led=25)
sd.detect(wait=True)

# Create a file and write something to it
#with open("/sd/test01.txt", "w") as file:
#    file.write("Hello, SD World!\r\n")
#    file.write("This is a test\r\n")

# Open the file we just created and read from it
with open("/sd/test01.txt", "r") as file:
    data = file.read()
    print(data)

`
working code:
`
from picographics import PicoGraphics, DISPLAY_TUFTY_2040, PEN_RGB565
from jpegdec import JPEG
import os
from time import sleep
import random

display = PicoGraphics(display=DISPLAY_TUFTY_2040, pen_type=PEN_RGB565)
j = JPEG(display)

directory = "images"

def random_shuffle(seq):
    l = len(seq)
    for i in range(l):
        j = random.randrange(l)
        seq[i], seq[j] = seq[j], seq[i]

files = os.listdir(directory)
os.chdir(directory)
while True:
  random_shuffle(files)
  for filename in files:
    if filename.endswith(".jpg"):
        print(filename)
        j.open_file(filename)
        j.decode()
        display.update()
        sleep(30)
`

problem code:
`
from picographics import PicoGraphics, DISPLAY_TUFTY_2040, PEN_RGB565, PEN_RGB332 
from jpegdec import JPEG
import uos as os
from time import sleep
import random
import machine
import sdcard

sd = sdcard.SDCard(0, 6, 7, 4, 5, led=25)
sd.detect(wait=True)

display = PicoGraphics(display=DISPLAY_TUFTY_2040, pen_type=PEN_RGB332 )
j = JPEG(display)

directory = "/sd/a"

def random_shuffle(seq):
    l = len(seq)
    for i in range(l):
        j = random.randrange(l)
        seq[i], seq[j] = seq[j], seq[i]

files = os.listdir(directory)
os.chdir(directory)
while True:
  random_shuffle(files)
  for filename in files:
    if filename.endswith(".jpg"):
        print(filename)
        j.open_file(filename)
        j.decode()
        display.update()
        sleep(30)

`

Error message:
`
MicroPython 67fac4e on 2023-02-16; Pimoroni Tufty 2040 with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
/sd Mounted
1.jpg
Traceback (most recent call last):
  File "<stdin>", line 33, in <module>
  File "/lib/sdcard.py", line 353, in readblocks
  File "/lib/sdcard.py", line 283, in readinto
OSError: Response Timeout

NB! Thonny could not execute 'Run'.

SCRIPT:

class __thonny_helper:
    import builtins
    try:
        import uos as os
    except builtins.ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = builtins.dict()
    @builtins.classmethod
    def print_repl_value(cls, obj):
        global _
        if obj is not None:
            cls.builtins.print('[object_link_for_thonny=%d]' % cls.builtins.id(obj), cls.builtins.repr(obj), '', sep='')
            _ = obj

    @builtins.classmethod
    def print_mgmt_value(cls, obj):
        cls.builtins.print('<thonny>', cls.builtins.repr(obj), '</thonny>', sep='', end='')

    @builtins.classmethod
    def repr(cls, obj):
        try:
            s = cls.builtins.repr(obj)
            if cls.builtins.len(s) > 50:
                s = s[:50] + "..."
            return s
        except cls.builtins.Exception as e:
            return "<could not serialize: " + __thonny_helper.builtins.str(e) + ">"

    @builtins.classmethod
    def listdir(cls, x):
        if cls.builtins.hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @builtins.classmethod
    def getcwd(cls):
        if cls.builtins.hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @builtins.classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @builtins.classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "<stdin>", line 3, in __thonny_helper
  File "/lib/sdcard.py", line 343, in readblocks
OSError: [Errno 5] EIO

You may need to reconnect, Stop/Restart or hard-reset your device.

`

It seems specific to 'j.decode()', as if this is commented out, the program executes normally.

Gadgetoid, 2023-02-28:
Is there a reason why you're not using the MicroPython lib's SDCard driver? https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py

We use this with many of our boards- which have onboard SD card slots- and have no issue displaying JPEGs from SD card.

quarterturn, 2023-02-28:
Does that library support the pico2040?

quarterturn, 2023-02-28:
It took a bit of searching to figure out how to init the SPI bus, but I did manage to get it working as such:

```
from picographics import PicoGraphics, DISPLAY_TUFTY_2040, PEN_RGB565, PEN_RGB332 
from jpegdec import JPEG
import uos as os
from time import sleep
import random
from machine import Pin, SPI
import sdcard

sd_spi = SPI(0, sck = Pin(6, Pin.OUT), mosi = Pin(7, Pin.OUT), miso = Pin(4, Pin.OUT))
sd = sdcard.SDCard(sd_spi, Pin(5, Pin.OUT))
vfs = os.VfsFat(sd)
os.mount(vfs, "/sd")

display = PicoGraphics(display=DISPLAY_TUFTY_2040, pen_type=PEN_RGB332 )
j = JPEG(display)

directory = "/sd/a"

def random_shuffle(seq):
    l = len(seq)
    for i in range(l):
        j = random.randrange(l)
        seq[i], seq[j] = seq[j], seq[i]

files = os.listdir(directory)
os.chdir(directory)
while True:
  random_shuffle(files)
  for filename in files:
    if filename.endswith(".jpg"):
        print(filename)
        j.open_file(filename)
        j.decode()
        display.update()
        sleep(30)

```

closed 2023-02-28 by quarterturn