pimoroni/pimoroni-pico #707: Question: Is there a way to attach a USB device to the Badger 2040 W?

TheTechRobo, 2023-03-06:
I have a Waveshare SIM HAT lying around. In addition to the HAT functionality, it can also be powered and communicated with over a USB connection. Does the Badger 2040 W support powering and communicating with USB devices? If not, does the Badger 2040 (non-W)?

Apologies if this is a duplicate, I didn't find anything.

alphanumeric007, 2023-03-13:
Communicate over UART maybe? You would have to solder the wires on to the PICO W though on a Badger W. On The Badger its the TX and RX solder pads. 
Micro Python, last time I checked, doesn't support USB HID. Circuit Python does though. I'm using it on my PICO RGB Keypad to send keycodes to my PC. I think you can do serial over USB though in Micro Python. Assuming that Hat can also work that way.

TheTechRobo, 2023-03-13:
Thanks, that's helpful! I assume the Pico's USB slot isn't powered? If it is, is it turned off when the Badger is halted (when software runs the halt method to enter the low-power state) or will it suck up a ton of power?

I'll look into UART. I think a makerspace recently opened up here in town, so I'll see if they have a soldering iron (and people that can use it, haha).

Thanks again!

alphanumeric007, 2023-03-14:
Ah, there is another wrinkle. On Battery Power, the PICO gets its power via VSYS. That gets you your 3.3V out but there won't be any +5V (VBUS) to feed the USB port. I completely forgot about that when I posted the above. Sorry about that, as it changes everything.

TheTechRobo, 2023-03-14:
Yeah, that's definitely an issue. I'm not sure there's much I can do about that. :/

helgibbons, 2023-04-05:
Thanks for diving in @alphanumeric007 !

Closing this one for now as hopefully you're now on the right track, but feel free to re-open if necessary.

closed 2023-04-05 by helgibbons