pimoroni/pimoroni-pico #718: Badger 2040 documentation confusing/incorrect about image formatting.

ConstantineK, 2023-03-18:


ConstantineK, 2023-03-18:
After reviewing the code more appropriate advice would be something like:

```
# On WSL:
sudo apt install imagemagick-6.q16

# Then run the following on your already resized 104x128 image
convert badge.jpg  -flatten -monochrome +compress badge_bit.jpeg
```
Upload the image with Thonny (you may have to select and open the folder in the UI within the badger itself if you want it to prompt you to upload it in the subfolder you are targeting)

Then update your badges/badge.txt file and change line 7 to the path of your image.

Reboot/reset the badger and your image will be presentable.

Gadgetoid, 2023-03-20:
Converting images is redundant since we added jpeg support. And Badger now lives here- https://github.com/pimoroni/badger2040

These old docs and tools probably need clearly deprecating sooner rather than later, but I had left them while the new Badger beds in.

helgibbons, 2023-04-05:
Hopefully the situation is a bit less confusing now - Badger 2040 (and Badger 2040 W) now have their [own repo](https://github.com/pimoroni/badger2040) with their own docs and I've updated the [Learn article](https://learn.pimoroni.com/article/getting-started-with-badger-2040) to describe the new way of doing things. The links on both shop pages are also now updated to point to the new repo.

Closing this one for now, but do feel free to [open a new issue over on the Badger 2040 repo](https://github.com/pimoroni/badger2040/issues) if there are documentation tweaks you'd still like us to make.

closed 2023-04-05 by helgibbons