pimoroni/pimoroni-pico #720: Pico Display 2.0 Button Test

rsrobbins, 2023-03-19:
The button test example for the Pico Display does not appear to work for the Pico Display 2.0. I changed DISPLAY_PICO_DISPLAY to DISPLAY_PICO_DISPLAY_2. None of the buttons work and nothing else happens.

helgibbons, 2023-03-20:
The code below seems to be working as expected for me using the latest release - which UF2 are you using (and what board are you using it on?)

```
# This example shows you a simple, non-interrupt way of reading Pico Display's buttons with a loop that checks to see if buttons are pressed.

import time
from pimoroni import Button
from picographics import PicoGraphics, DISPLAY_PICO_DISPLAY_2, PEN_P4

# We're only using a few colours so we can use a 4 bit/16 colour palette and save RAM!
display = PicoGraphics(display=DISPLAY_PICO_DISPLAY_2, pen_type=PEN_P4, rotate=0)

display.set_backlight(0.5)
display.set_font("bitmap8")

button_a = Button(12)
button_b = Button(13)
button_x = Button(14)
button_y = Button(15)

WHITE = display.create_pen(255, 255, 255)
BLACK = display.create_pen(0, 0, 0)
CYAN = display.create_pen(0, 255, 255)
MAGENTA = display.create_pen(255, 0, 255)
YELLOW = display.create_pen(255, 255, 0)
GREEN = display.create_pen(0, 255, 0)

# sets up a handy function we can call to clear the screen
def clear():
    display.set_pen(BLACK)
    display.clear()
    display.update()

# set up
clear()

while True:
    if button_a.read():                                   # if a button press is detected then...
        clear()                                           # clear to black
        display.set_pen(WHITE)                            # change the pen colour
        display.text("Button A pressed", 10, 10, 240, 4)  # display some text on the screen
        display.update()                                  # update the display
        time.sleep(1)                                     # pause for a sec
        clear()                                           # clear to black again
    elif button_b.read():
        clear()
        display.set_pen(CYAN)
        display.text("Button B pressed", 10, 10, 240, 4)
        display.update()
        time.sleep(1)
        clear()
    elif button_x.read():
        clear()
        display.set_pen(MAGENTA)
        display.text("Button X pressed", 10, 10, 240, 4)
        display.update()
        time.sleep(1)
        clear()
    elif button_y.read():
        clear()
        display.set_pen(YELLOW)
        display.text("Button Y pressed", 10, 10, 240, 4)
        display.update()
        time.sleep(1)
        clear()
    else:
        display.set_pen(GREEN)
        display.text("Press any button!", 10, 10, 240, 4)
        display.update()
    time.sleep(0.1)  # this number is how frequently the Pico checks for button presses
```

helgibbons, 2023-03-20:
Hmm, I wonder if trying to `update()` the screen right at the beginning of the program is causing problems - the example is working for me but I did encounter a few MicroPython hardlocks that required a reset.

Does adding a short sleep before triggering the clear function (and invoking the first `display.update()`) help?

```
# set up
time.sleep(0.5)  # short pause to allow USB time to initialise
clear()
```

Gadgetoid, 2023-03-23:
There are some monumental changes in the 1.19.17 which could cause weirdness. If you can find a minimal failure case I can do some sleuthing.

The example right from GitHub seems to run fine on a *Pico* at least.

Gadgetoid, 2023-03-23:
Okay I think I'm barking up the wrong tree here, `PEN_P8` - which also requires converting from a palette mode to `RGB565` - appears to work...

helgibbons, 2023-04-05:
This should be fixed in v1.19.18 so closing this one for now - let us know if you're still having trouble though!

closed 2023-04-05 by helgibbons