pimoroni/pimoroni-pico #721: badger2040w clock example time zone offset?

pketelaar, 2023-03-21:
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/badger2040w/examples/clock.py 

Doesn't include anything for a time zone offset, manually putting in an hour adjustment will not work. Can you update with the code from the non-wireless version to adjust the time to local?

Gadgetoid, 2023-03-21:
This code has been deprecated and Badger moved over to https://github.com/pimoroni/badger2040

However, your issue is still valid. It looks like the canonical way to apply a timezone offset in this case would be to get the correct time in the first place from the ntp server, since `ntptime.settime()` accepts an offset value: https://mpython.readthedocs.io/en/master/library/micropython/ntptime.html

So you can just change the line `ntptime.settime()` to `ntptime.settime(timezone=HOURS)` where `HOURS` is your desired offset (the default is 8), and replug the battery (give it a few seconds to discharge) so it syncs the time again.

Gadgetoid, 2023-03-21:
I've made some changes to clock in the new repo, including bringing back the time set functionality (which persists on a Badger W even if Thonny tries to reset the time) and surfacing a `TIME_OFFSET` value which I might have to do more with at some point. (Maybe making it part of WIFI_CONFIG)

Test builds and discussion here: https://github.com/pimoroni/badger2040/pull/2

closed 2023-03-21 by Gadgetoid

pketelaar, 2023-03-21:
I tried that but getting the following error:

import time
import machine
import ntptime
import badger2040

display = badger2040.Badger2040()
display.set_update_speed(2)
display.set_thickness(4)

WIDTH, HEIGHT = display.get_bounds()

try:
    display.connect()
    if display.isconnected():
        ntptime.settime(timezone=-4)
except (RuntimeError, OSError):
    pass  # no WiFI

rtc = machine.RTC()

display.set_font("sans")

———

>>> %Run -c $EDITOR_CONTENT
Client True 192.168.7.141
Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
TypeError: unexpected keyword argument 'timezone'

Gadgetoid, 2023-03-22:
Oh good grief. What on earth are these docs I’ve linked to? Sorry, you’re right; I’m barking up the wrong tree here.

Nonetheless the next release of Badger 2040 should help by giving you the manual time setting back. Bear with me! And watch this space: https://github.com/pimoroni/badger2040