pimoroni/pimoroni-pico #724: Buttons not available on Pico Unicorn 1.19.17

NHurst99, 2023-03-23:
Pico Unicorn Pack does not have any button variables in 1.19.17
[image]
If it still is available, the unicorn examples should be updated, as the example code is what I used.

Buttons were working in 1.19.16:
[image]

Gadgetoid, 2023-03-23:
The button constants are on the module, rather than the class instance.

Example code needs to reflect this!

Gadgetoid, 2023-03-27:
Added a commit onto https://github.com/pimoroni/pimoroni-pico/pull/733 to change the module, rather than the examples. Should be in the next release!

closed 2023-03-27 by Gadgetoid