pimoroni/pimoroni-pico #740: IPS Display, DS3231 and LIS3DH

condor13, 2023-03-30:
Hey everybody, I`m trying to make a portable watch with this 3 bad boys :)) (the pimoroni ips display, ds3231 (rtc as I dont want to use picoW) and a lis3dh (simple giro, for tap and rotation sensor))
I have no problem setting up the rtc and gyro (both on I2C on bus 1) but when I add the display the pico is frozen. If I use the display and gyro is the same issue.  
As I understand from schematic the display is on SPI bus. 
I tried lowering the sampling rate on the gyro but with no luck. 
Let me know if you have any ideas :) 
Used the library from Piico 3-Axis Accelerometer LIS3DH 

Thanks 

[image]
[image]
[image]

simonhenry700, 2023-03-30:
What firmware are you using? 1.19.17 (Pimoroni Pico or PicoW variant) has a known problem - supposedly fixed in 1.19.18 (but I haven't tested that yet). I do know that 1.19.16 works with the PicoDisplay.

Simon

condor13, 2023-03-30:
@simonhenry700 haha...missed that thread :) 1.19.17. Upgrading now. Thanks

condor13, 2023-03-30:
Nope. That didn't fix it.

simonhenry700, 2023-03-30:
Have you tried 1.19.16?

condor13, 2023-03-31:
@simonhenry700 Yes. Got it running in 1.19.18 with a simple test: text display of axis orientation and clock to see if all I2C devices can work together. But as soon as I add graphics it crashes... I give up the gyro and use the buttons to wake the display backlight.

simonhenry700, 2023-03-31:
OK - I'm very new to this so I'm not likely to be able to help you any further. Maybe you need to report this as a potential bug. Good luck, Simon.

condor13, 2023-03-31:
@simonhenry700 yeah me too. I`m new too. to much oh a hassle... wanted to add the gyro for tapping to wake the device. Anyway thanks for the idea.

Gadgetoid, 2023-03-31:
Could you share any code, at all, or the minimum amount required to trigger the freeze? This shouldn't happen, but it'll no doubt be tricky to track down.

condor13, 2023-04-01:
@Gadgetoid Hello. sorry for my messy code. 
https://github.com/condor13/pico-solar-system
Here is the code. 
Line 64 when I try to acces the TAP function it freezes. 
I think this is from my sloppy progaming skills to combine 2 projects.
Line 190-196 its my dumb way to make a countdown to turn-off the backlight.

condor13, 2023-04-24:
> Could you share any code, at all, or the minimum amount required to trigger the freeze? This shouldn't happen, but it'll no doubt be tricky to track down.

No? Anything?

condor13, 2023-05-17:
Thanks. Please close :)

Gadgetoid, 2023-05-17:
Sorry, getting pulled in all directions. Did you get anywhere?

Unfortunately the LIS3DH driver is code I'm unfamiliar with it and I can't see any obvious reason why it would cause a lockup.

Might be worth trying `PEN_P8` instead of `PEN_RGB565` to see if it's RAM related. I think PicoGraphics will handle the colours appropriately.

Also worth trying v1.20.2 for what difference that might make: https://github.com/pimoroni/pimoroni-pico/releases/tag/v1.20.2

closed 2025-01-08 by thirdr