pimoroni/pimoroni-pico #741: Ghosting on Cosmic Unicorn

Leone25, 2023-03-31:
Hello,

while working on a project on my Cosmic Unicorn I noticed that there is a ghost image shifted down one line of what is displayed on the screen, plus the last line is "ghosted" in the middle of the screen.
I've already contacted support about this and they were able to reproduce the issue as well so it seems like it's a firmware issue.
Here is a picture of what it looks like with [this code](https://pastebin.com/wWpyMRHf) code running (taken in a dim room with phone in night mode): 
[image]

This seems an issue very similar to #666 .

I hope it's possible to fix.

Thanks in advance

Enrico

Gadgetoid, 2023-03-31:
@MichaelBell figured it out for Galactic and the same solution should be portable, as far as I'm aware. It was on my TODO list, but I haven't wrapped my brain around the code yet.

Leone25, 2023-03-31:
great, will wait for a fix
thx

MichaelBell, 2023-04-26:
Possible fix on this branch: https://github.com/MichaelBell/pimoroni-pico/tree/cosmic-ghost-fix
This was just applying the same thing I did on galactic, but I'm away from home so haven't tested it - can do that in a couple of weeks!

MichaelBell, 2023-05-10:
I've had a chance to test this now - looks like that fix does work so I've submitted a PR.

closed 2023-08-08 by helgibbons