pimoroni/pimoroni-pico #743: Pico Explorer nothing works

codaysworld, 2023-04-05:
Have installed pimoroni-pico-v1.19.18-micropython.uf2 on my Pico Pi, have downloaded Thonny and nothing works. 2 Days after, downloading libraries in the terminal, reading a few of instructions, watch tutorials, the lcd display is black, no demo file runs, errors, don't find this, that ... 

the only thing ist to lighten up the led. use a Mac. what happened?

thanks

alphanumeric007, 2023-04-07:
Is Thonny set to the correct interpreter in the lower right hand corner? 
If it's not, it thinks your running Python, not Micro Python, etc.

helgibbons, 2023-04-26:
Closing this for now, but feel free to re-open (or drop us an email at support@pimoroni.com ) if you're still having trouble!

closed 2023-04-26 by helgibbons